pftf/RPi4 #202: Installing OS on the same drive that UEFI

ultraSsak, 2021-12-08:
Hi there,
Please excuse my newbie'sh question, but here it is: 
Lets say i have MicroSD card with plethora of space.
I create one small partition for UEFI.
Is it possible to install the OS on rest of the space of this MicroSD?

I tried this with booted from pendrive Alma Linux, but it didn't see the card as disk device (to partition)

jlinton, 2021-12-09:
Yes, this is one of the cases I test (uefi+fedora on SD).  The key is making sure your OS doesn't destroy the fat/ESP partition that the firmware resides on during the install. This is usually the same process as installing a multiboot system.

As far as Alma Linux, its going to boot/etc, but you're not going to have drivers for the onboard ethernet, wifi, sd, etc until they release something closer to centos stream 9. That is simply because of the fact that the drivers weren't merged into 4.18, and no one at RH/etc has backported them because it's not an "enterprise" platform. So, as you noticed it's possible to boot/use it with alma you're going to need to stick to USB3 devices that have drivers built-in for networking/storage.

closed 2021-12-09 by jlinton

ultraSsak, 2021-12-09:
Thanks for comprehensive explenation.