pftf/RPi4 #203: question: is it feasible to execute a UEFI-BIOS using x86 instructions on non-x86 platforms?

mi-hol, 2022-01-24:
RAID disc controller cards have usually a build-in configuration program in their BIOS. [Some cards added a UEFI-BIOS](http://www.dawicontrol.com/manuals/dc-raid_en.pdf) but still only build this for x86 platforms. 

I wonder if it is feasible to execute this code on the growing number of non-x86 platforms via some emulator (qemu,..) during boot.

sunnywang-arm, 2022-01-26:
@mi-hol 
I think you should bring your question to devel@edk2.groups.io. This is not related to Rpi4. Rpi4 even doesn't have a PCIe slot for installing the RAID card.  Could you close this issue?  By the way, I'm not clear about your question. but it looks like related to EFIAPI.

mi-hol, 2022-01-26:
@sunnywang-arm 
Thanks for the suggestion, I'll try it.

> This is not related to Rpi4. Rpi4 even doesn't have a PCIe slot for installing the RAID card

[RPI4 Compute Module (CM4](https://www.raspberrypi.com/products/compute-module-4/?variant=raspberry-pi-cm4001000)) put inside the official IO board actually has a PCIe slot and [here is a blog post](https://www.jeffgeerling.com/blog/2021/hardware-raid-on-raspberry-pi-cm4) using an enterprise grade Broadcom/LSI RAID controller.
The author of post is maintaining[ this complete web site ](https://pipci.jeffgeerling.com)dedicated to publish test results for PCie cards in CM4 powered boards. Hope this helps.

mi-hol, 2022-01-27:
replies to my question from devel@edk2.groups.io

Andrew Fish 7:57am   [#86111](https://edk2.groups.io/g/devel/message/86111)  
https://www.suse.com/c/revolutionizing-arm-technology-x86_64-option-rom-aarch64/

https://www.workofard.com/2020/12/aarch64-option-roms-for-amd-gpus/

0n0w1c, 2022-01-28:
This is very interesting indeed.

jlinton, 2022-01-31:
Actually there is already an x86 option ROM emulator in edk2. Its used for PXE booting on various PCIe cards, as well as bootstrapping AMD/Nvidia graphics cards. Its fairly trivial to enable on the rpi4/cm4, but I haven't bothered. If you look at the socionext and solidrun platforms you will see the build flag needed to automatically enable it.

(although I guess now that I've typed all this you got a response about it on edk2)

Ideally we would figure out a way to only enable it for CM4, but at the same time, that would break our "one firmware for all pi4 variants" plan.

mi-hol, 2022-02-17:
> If you look at the socionext and solidrun platforms you will see the build flag needed to automatically enable it.

@jlinton as I'm completely new to building low level code I'm unsure if I found the correct reference.
Is it in https://github.com/tianocore/edk2-platforms/tree/63d520f9431a6aa4ce1df4387dd0a9831f8a13b8/Platform/SolidRun/Armada80x0McBin#build-edkii
the parameter "-D X64EMU_ENABLE"?

mi-hol, 2022-02-20:
in a [related issue here ](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/132#issuecomment-1004225784)[rtreffer](https://github.com/rtreffer) wrote:
"You will need an UEFI compiled with DEFINE X64EMU_ENABLE = TRUE (add it to the [rpi dsc file](https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/RPi4/RPi4.dsc))
This will add a Qemu to your UEFI. Loading these drivers will be slow, borderline slow, but it will work
**You will need a x86 machine with the controller and all disks to initialize the RAID. I didn't get the classic POST to run**"

so still there seems to be a road block

paulwratt, 2022-02-21:
> You will need a x86 machine with the controller and all disks to initialize the RAID

Has anyone had a look at the Videos of Hardware Raid used with the CM4 based on Jeff Geerlings PCIe repository? I know for a fact that he had to deal with a couple of cards that have x86 Bios - there maybe more insight into this by reviewing some of them.

pbatard, 2022-03-07:
While it might be possible to virtualize x86 execution, I would say that this generic question is out of scope for this project, so I will close this issue.