pftf/RPi4 #204: Adafruit PiRTC DS3231

zap8600, 2022-01-26:
Hi. I'm making a laptop with my Pi 4 and I've decided to run Windows on it. I'm going to add a [Adafruit PiRTC DS3231](https://www.adafruit.com/product/4282) and I'm wondering if adding `dtoverlay=i2c-trc,ds3231` to the `config.txt` will work. I thought I would need to code some Windows drivers (which I wouldn't mind), but if it is supported, it would make things a lot easier. Will it work?

jlinton, 2022-01-31:
Oh another RTC, this isn't really a device tree thing as UEFI has runtime support for RTC's that should be the primary enablement path (for the time being). If you want to write a windows driver, then you need to add it to the DSDT when the device is detected. 

Basically, step one is just to enable it for edk2, and it should work on all OS's without a driver.

zap8600, 2022-01-31:
So, the RTC is supported? If so, then do I still need to add the `dtoverlay` to `config.txt`?

jlinton, 2022-02-01:
No, I started implementing the RTC, on the official CM4 IO board, but never finished. I guess we need another "stuff to finish" list. The point though is that you only strictly need it to work in UEFI because the existing firmware has the framework in place to support the UEFI RTC runtime services used by most OS's. So, DT doesn't need to get involved. I have another one of the inexpensive boards like that one you point out around here also on my own personal TODO, but if someone else gets to it first, great!

paulwratt, 2022-02-17:
@zap8600 make sure to post back with progress/fix/patch or repo link.

jlinton, 2022-06-02:
I think that thing might be compatible with the Ds1307, which has a driver already, and is have working).

So see if this branch helps any: https://github.com/jlinton/edk2-platforms/tree/cppc6-various-fixes-vc4aml-mailboxspinlock-xhcipci-clocktweaks-spiv5

asunekants, 2023-04-02:
@jlinton - Not a dev here, but really interested in getting the RTC to work on RPI4; How would one go about building a firmware using your branch?
Thanks :)

jlinton, 2023-04-03:
There are usually binary images checked into those branches ex: https://github.com/jlinton/edk2-platforms/blob/cppc6-various-fixes-vc4aml-mailboxspinlock-xhcipci-clocktweaks-spiv6-FPDT-GPUHACKING/RPI_EFI.fd

But they can also be built with the standard edk2 build instructions. I have another repo https://github.com/jlinton/rpi_efi_build which has a couple of shell scripts that checkout and build TFA and EDK2 if your running on fedora.

At some point I'm actually going to start upstreaming that stuff, but I have some GPU hacking that eats my rpi4 hacking time.

andreamtp, 2023-12-29:
Hi @jlinton have you succeeded in your goal of upstreaming your contributions for addint RTC support for tianocore on rpi?

andreamtp, 2023-12-29:
Hi @jlinton I'm unsure if it's the latest release, but I've tested this build from you:

https://github.com/jlinton/edk2-platforms/blob/cppc8/RPI_EFI.fd

and my DV3231 has been correctly used by tianocore as RTC, in fact my system booted like a charm:
~~~
[    0.271758] rtc-efi rtc-efi.0: registered as rtc0
[    0.276862] rtc-efi rtc-efi.0: setting system clock to 2023-12-29T16:30:12 UTC (1703867412)
~~~

Any chance to add a PR here or upstream to make it available to everybody?

jlinton, 2024-01-03:
Yes, I've been cleaning up this and a few other commits in that repo for upstreaming last month, but got stalled debugging a shim/etc problem. Once that is cleared up I will start posting the set.

andreamtp, 2024-02-22:
Hi @jlinton thanks for your effort: is there any progress about the upstreaming of your code?

Roeda, 2024-04-06:
> Yes, I've been cleaning up this and a few other commits in that repo for upstreaming last month, but got stalled debugging a shim/etc problem. Once that is cleared up I will start posting the set. Which should be late next week, assuming everything goes a planned.

indeed I've been looking for a solution like yours for sometime, as Coreos (ignition file based) is a mess without a proper RTC.
any recommendation on how to use your branch, or is it better to wait for your upstreaming ? (I shall first quickly buy a DS1307)

Roeda, 2024-04-06:
> Hi @jlinton I'm unsure if it's the latest release, but I've tested this build from you:
> 
> https://github.com/jlinton/edk2-platforms/blob/cppc8/RPI_EFI.fd
> 
> and my DV3231 has been correctly used by tianocore as RTC! My system booted like a charm:
> 
> ```
> [    0.271758] rtc-efi rtc-efi.0: registered as rtc0
> [    0.276862] rtc-efi rtc-efi.0: setting system clock to 2023-12-29T16:30:12 UTC (1703867412)
> ```
> 
> 👏 thank you!
> 
> Any chance to add a PR here or upstream to make it available to everybody?

@andreamtp could you please share your built uefi image for rpi4 that has the RTC working ?

andreamtp, 2024-08-09:
> @andreamtp could you please share your built uefi image for rpi4 that has the RTC working ?

@Roeda I'm using the build from Jeremy: download the blob from here:
https://github.com/jlinton/edk2-platforms/blob/cppc8/RPI_EFI.fd

Prepare the boot device with the latest release from the official project, then overwrite the existing RPI_EFI.fd with the one from Jeremy: at boot you'll find the new options.

@jlinton what is the status of upstreaming your great work? Would be really amazing to have your improvements available to everybody.

Thanks