pftf/RPi4 #205: Low fps on RPi4

nebiyebulan, 2022-02-03:
I am running my project with yolov5 on ubuntu operating system on RPi4. Normally when I measure fps on webcam, it gives very good results. but it runs my object detection project very slowly. It shows fps as "0".

[image]

TheMindVirus, 2022-02-06:
In the Windows Driver, disconnecting the interrupt for V-Sync seemed to help increase Internal Frames Per Second
but at the cost of Screen-Tearing when moving things around on-screen or playing back video.

In Linux, there are several commands and environment variables to turn off V-Sync etc.,
such as an `xrandr` video mode with `-vsync` and the solution suggested here: `vblank_mode=0 <app>` to launch an \<app\>
https://stackoverflow.com/questions/17196117/disable-vertical-sync-for-glxgears

Vertical Sync would be fairly low priority on a Pi4 for most Machine Learning Development tasks,
but there are other Display Settings to check if having V-Sync switched on is important to you and your project.

pbatard, 2022-03-07:
I believe this is an Ubuntu specific question rather than a UEFI one. As such I will close this issue.

closed 2022-03-07 by pbatard