pftf/RPi4 #206: Website link is dead

barcharcraz, 2022-02-09:
Well not "dead" but rpi4-uefi.dev has an HSTS policy and the cert has expired, so it's inaccessible in most browsers (you can bypass the warning, sort of)

jlinton, 2022-02-11:
Yah, I noticed that too, I pinged the responsible parties.

tux2bsd, 2022-02-11:
The SSL certificate has expired.  Someone wasn't checking their emails.

Move to Let's Encrypt.  (note, I didn't say to do this as an immediate solution. Though doing so is quick...)

andreiw, 2022-02-11:
Yes. I did in fact renew, but what I missed is the yak shaving process
involved to "install" the certificate. Needlessly, senselessly
overcomplicated garbage. Taking care of it now.

jlinton, 2022-02-17:
Closing because it appears to be working now.

Thanks.

closed 2022-02-17 by jlinton

WinkelCode, 2023-06-23:
It’s down again due to having no DNS applicable DNS record.