pftf/RPi4 #210: Make UEFI variables stored outside of RPI_EFI.fd

hellosway, 2022-03-09:
When I update, it makes all configuration be lost, this is very concerning on headless servers because the uefi interface cannot be assessed. If it is stored in a separate file, for example RPI_EFI_VARS.fd then it can be excluded from an update, by package managers or manually.

jlinton, 2022-03-10:
See if https://github.com/jlinton/edk2-platforms/blob/cppc6-various-fixes-vc4aml-mailboxspinlock-xhcipci-clocktweaks-spiv4/RPI_EFI.fd fixes your problem.

rgl, 2023-05-19:
Any update on this?

Having the variables in another file also opens the door for using https://gitlab.com/kraxel/virt-firmware to manipulate the VARS file without the need to boot a RPi, as it can be manipulated in another machine before writing it into the final sdcard.