pftf/RPi4 #211: How to modify boot screen？

yyds-yyds-yyds, 2022-03-12:
Hello, I want to ask a question, how to modify the boot screen? I want to change the boot screen, but I don't know how to change it. Can you tell me how to change it?

hellosway, 2022-03-14:
What exactly do you mean by boot screen. Is it the raspberry pi logo with the progress bar at the bottom.

yyds-yyds-yyds, 2022-04-08:
Yes, it's the logo. How to change it! Sorry, my English is poor and difficult to describe clearly. Thank you for reading patiently.

TheMindVirus, 2022-04-08:
https://github.com/tianocore/edk2-non-osi/blob/master/Platform/RaspberryPi/Drivers/LogoDxe/Logo.bmp

yyds-yyds-yyds, 2022-05-02:
So the boot image cannot be changed?

TheMindVirus, 2022-05-02:
It can be changed but UEFI in its current state needs recompiling after you change it in the repository. I have changed mine in this way.

In the RPI_EFI.fd archive itself it seems to have been turned into a binary stream which is mangled with other data inside LogoDxe.efi instead of being an editable .bmp file.

pbatard, 2023-06-29:
Changing the boot logo requires editing https://github.com/tianocore/edk2-non-osi/blob/master/Platform/RaspberryPi/Drivers/LogoDxe/Logo.bmp as @TheMindVirus pointed out and recompiling your own UEFI Firmware. We do not support trying to edit the logo in the `RPI_EFI.fd` that is provided from this repository.

closed 2023-06-29 by pbatard