pftf/RPi4 #213: Question: is it possible to upgrade the UEFI firmware?

arajczy, 2022-03-23:
Hello,

Is it possible / recommended to upgrade the UEFI firmware on an already installed system? If yes then what are the required steps?

I have a Raspberry Pi 4 Model B with 8 GB RAM installed Fedora 35 with  UEFI Firmware v1.30.

Thanks

TheMindVirus, 2022-03-23:
For a system running Ubuntu 1804, manually follow the steps that the Continuous Integration (CI) Build follows to rebuild it
after it has been tweaked and upgraded: https://github.com/TheMindVirus/pftf-rpi4/blob/main/appveyor.yml
The process will be similar for Fedora but might have differences in some of the commands.

Once it has built it will create RPI_EFI.fd (or similar .fd file) located within one of the subdirectories of the Build folder.
Once you have a replacement RPI_EFI.fd to replace Firmware v1.30, rename it to something else to back it up
and then paste in the upgraded RPI_EFI.fd into the boot partition (or whichever file config.txt points to).

It is recommended to create a custom fork of this repository to track any changes you've made to your version of the firmware.

jlinton, 2022-03-25:
I'm not sure why any of that would be needed, just unzip the newer version of the binaries from this project over top of the older version. You can do this from the rpi itself. 

Then on first boot the efi fallback should reconfigure the missing boot variables. If you changed any of the EFI settings those will need to be reset as well (the 3G limit flag for example).

Course backing up the old version as mentioned above might be helpful too if you happen to need to revert it.