pftf/RPi4 #215: cannot change 3GB limit, setvar reports wrong value

mach0gr, 2022-04-23:
Hey all, I'm yet another guy trying to install esxi fling on RPi4 8GB (have successfully done it few months ago but didn't keep it)
I have updated the pi EEPROM 
`BOOTLOADER: up to date
   CURRENT: Tue 25 Jan 14:30:41 UTC 2022 (1643121041)
    LATEST: Tue 25 Jan 14:30:41 UTC 2022 (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1`
and trying with your new release UEFI Firmware v1.33 I have stumbled across this issue.
I noticed that I when I disabled the RamLimit of 3GB required for the esxi fling to install, the change did not persist after reboot. 
Going to `Device Manager` → `Raspberry Pi Configuration` → `Advanced Settings` make change, save, reboot and when I go back to UEFI UI it still shows 3GB...

Out of curiosity I entered the Shell (F1) on boot and tried to read/change the RamLimitTo3GB as stated in [https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi4](url) 
So when I enter the shell and type
`setvar RamLimitTo3GB -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5` to read the value, I get a value of `01 00 00 00` which is wrong... The expected values are 
Disable = 0x00000000
Enabled= 0x00000001 (default)

I change the value with 
`setvar RamLimitTo3GB -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5 -bs -rt -nv =0x00000000`
I read it back which shows up fine, but when I reboot it seems it didn't persist and reads back at `01 00 00 00`

I'm I doing something wrong ? Any help appreciated

mach0gr, 2022-05-28:
OK, tracked down the issue to be that "write" or "save" settings I made in UEFI UI did not persist or actually being "saved" on the SD card I was using which a really old microSD 512MB.
I tried with a newer microSD 64G, same files, all works perfectly. I then copied everything to my old SD card and everything works/boots but of course I cannot save any modifications (but I don't care at the moment).

I see many other people having issues with changes in UEFI UI settings not persisting I'm not aware of any SD limitation (developers here may want to track this or not) I would suggest just try with a new SD :)

pbatard, 2023-06-29:
Considering that you managed to solve/work around your problem, I will close this issue.

closed 2023-06-29 by pbatard