pftf/RPi4 #217: Unified Kernel Images failed to boot.

hellosway, 2022-05-08:
Unified kernel images by from following the manual section from https://wiki.archlinux.org/title/Unified_kernel_image do not boot. They get the error `Image Failed To Load`, but it should work. This occurs on archlinuxarm, and I used the aarch64 stub. I have not tried any other distros. I used systemd-boot as the bootloader.

jlinton, 2022-06-02:
Hmm, what is the error? Can you attach the earlycon/etc output here? Because this should be working (and has in the past) if you just start it from the uefi shell.

jlinton, 2022-06-03:
Their instructions differ a bit from the way I have been doing it. With arm64 you should generally ignore the bits about appending the kernel image, and either compile with a default 'initrd=someimagefile' appended to the kernel command line, to start it via something that can pass parameters (which generally isn't uefi, but grub can). Then assure that the somefine initrd image is on the ESP along with the kernel.

sharpenedblade, 2022-06-10:
EFI stubbed images do work, the default arch and fedora aarch64 kernel images boot and are EFI-stubbed, only booting with the initramfs and command line bundled fails. Maybe systemd-stub is broken with this build, it works with a x86 tianocore/edk2 uefi. I think it might have to do with the magic 0x30000 and similar numbers used in objcopy.

jlinton, 2022-06-14:
Ok, couple of comments, what does `file combinedvminitrd.efi` say?

Also, are you using the systemd-stub command to create the image or that ugly objcopy/etc sequence you linked? I would try systemd-stub, although I don't seem to have it on my fedora install. Instead I just tried using systemd-boot itself, and that all seems to be working fine. So, it does boot with something other than grub and kernel/efi stubbed images.

sharpenedblade, 2022-06-19:
The objcopy command is currently the only way to build the images, everything else uses the same command, it was originally found by a systemd dev I think. systemd-stub is part of an binary, you dont run it, it is included in a efi image to properly load the kernel. All bootloaders I have tried work, but nothing works with an efistubbed kernel image. Fedora does not use systemd-stub or efi images, they sign the kernel so it cannot be combined into a efi image, the signature would fail if it was.

zhanghua000, 2023-04-30:
If you are using `Image.gz` instead `Image` to build UKI Image, this cannot be solved currently because systemd-stub cannot decompress `Image.gz`. The compressed Kernel Image can decompress itself on x86_64 while can't do the same on aarch64, the compressed aarch64 kernel plans to let boot loader to do the decompression job[^1]. So maybe the solution is that let aarch64 version of systemd-stub decompress the compressed Image[^2].
[^1]: https://www.kernel.org/doc/html/latest/arm64/booting.html#decompress-the-kernel-image
[^2]: https://github.com/systemd/systemd/issues/23788