pftf/RPi4 #218: Can not boot ISO live disk

linuxgnuru, 2022-05-13:
When attempting to boot an arm ISO live disk with pftf I get:
```
EFI stub: Booting Linux Kernel...
EFI stub: Generating empty DTB
EFI stub: Loaded initrd from command line option
EFI stub: Exiting boot services...
```
then it just hangs with no other messages. If, however, I use tow-boot the same USB flash live disk boots without any problems

jlinton, 2022-06-02:
Can you put earlycon on the boot command line and attach the output. Which ISO is this?

linuxgnuru, 2022-06-03:
It is an ISO of Pop 22.04. Using tow-boot it was able to boot.  I'll add the earlycon and post the output in a bit.

ofthesun9, 2022-06-24:
When upgrading to ubuntu 22.04 (vmlinuz-5.15.0-37-generic), I noticed that I lost the video output (booting with ACPI only).
I still can login, but this is an headless set-up right now :-)
I guess there is an issue with the framebuffer initialization ?

linuxgnuru, 2022-06-28:
Sorry it's taken me awhile to get back, I work at system76 which is how  I got the ISO.  We're still working on things to and once they are ironed out I'll be able to supply more info.