pftf/RPi4 #224: Add hysteresis to ACPI fan control

dogtopus, 2022-12-04:
Unlike what you can do in device tree, the temperature threshold for ACPI fan doesn't seem to have hysteresis. This causes the fans to ramp up and down very frequently and create unwanted noise.

jlinton, 2022-12-14:
Presumably, your just using the GPIO fan? In Linux?

This is an issue for the older ACPI on/off methods in linux, where the ACPI expectation is that the OS has hysteresis (and of course linux expects newer variable speed fan methods/firmware support). I have a similar problem with mine where the heatsink and fan are way overspec'ed, and it cycles on, and can pull the temp down 5C+ within a few seconds, and then cycles off. My solution is either to adjust the setpoint in the firmware or use one of the 3rd party scripts that controls it manually, adjust the setpoint with the kernel tools tmon/etc. Usually down so the fan turns on sooner. Or in my primary use case I just put an itty bitty fan on it that moves a lot less air.

But of course with tmon (and /sys/class/thermal/thermal_zone*) you will note there is a PID controller being exported by linux, but it doesn't appear to work with this fan, and that maybe is the TODO here as I'm not really sure why. I suspect without looking at the code again its only hooked up for the variable speed support (but I don't think it was working on my downstream poe+fan hat patches either).

You might poke someone on a kernel ML about why the older ACPI methods still aren't hooked up to the PID controller by default.

dogtopus, 2023-02-26:
I'm not sure why this has to do with the kernel. The threshold is set on the UEFI setup screen, and there's only a threshold value, not a range that you can specify.