pftf/RPi4 #228: EFI shell/bootup non-responsive over serial

ddevault, 2022-12-23:
I get logs and other debug info printed on the serial port, but attempting to interact with the system over serial does not work. When prompted to press a key (Esc, F1, or Enter) to choose between config, shell, or boot, pressing it over serial has no effect. When launching the EFI shell with the keyboard, the shell output appears over serial but cannot be interacted with. Typing commands into the keyboard causes them to appear over serial.

pbatard, 2022-12-24:
I am not able to replicate your issue with firmware 1.34.

This looks like cabling issue more than anything else, because what you describe is what typically happens if you have the Tx pin or wire disconnected.

I would strongly suggest that you test your serial connection outside of the UEFI firmware (by installing the default Linux Raspberry Pi distribution and see if you can send serial input then).

Considering that I am unable to replicate your problem, which indicates that this is not an issue with the UEFI Firmware, and that nobody else is reporting a similar problem, I will close this issue.

closed 2022-12-24 by pbatard

ddevault, 2023-01-07:
Able to reproduce with a different serial cable and a different rpi. Have not attempted to repro from Linux yet, though.

ddevault, 2023-01-18:
Finally actually needed serial to work and got to the bottom of this. Hardware flow control was enabled on the host but disabled on the pi.