pftf/RPi4 #230: no cpufreq with debain 11

chriss-de, 2023-02-06:
As the title says . 

I'm using Debain 11 and this EFI on a RPi4b 8GB. The system is running with the default freq you can set in the EFI und RPi Configuration > CPU
The OS cpufreq driver does not detect a CPU that it can work with? I'm not really sure who is to blame - maybe just myself :D

I searched the internet up and down but could just find one other guy who mentioned/asked it.

---
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 1:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 2:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 3:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
```

TheMindVirus, 2023-02-06:
I'm just going to leave this here, mumble mumble: https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#set-clock-rate

Part of it is due to Pi4 UEFI not being a full emulation, just the essential changes required to boot custom built OS versions.
I tested this mechanism and it didn't seem to actually change the CPU frequency for all clocks, only some of them.
There are differences between Pi3 and Pi4 that prevent existing code from completing the request successfully and reliably.

Chances are that the Linux kernel source used to compile your Debian 11 image does not have the changes to this mechanism.
A custom version of either the Linux kernel has to be rebuilt or every piece of software has to have an edge case workaround for every slight hardware reconfiguration and mechanical difference that exists between different versions of the platform.

I would be keen to know if this issue also affects Raspberry Pi OS running on UEFI vs. Standalone, Hardware vs. Virtual Machine.

chriss-de, 2023-02-09:
Thank you for that insight. I'm closing the issue for me.

closed 2023-02-09 by chriss-de