pftf/RPi4 #231: Boot next device in case PXE failure

Banshee1221, 2023-02-24:
Hi there,

I'm really interested in using this project to provision Raspberry Pi nodes via the [Ironic ](https://docs.openstack.org/ironic/latest/) bare-metal provisioner. I've managed to make some progress with it in that I'm able to PXE boot the node and provision the disk image onto a USB attached SSD, but I've noticed 2 issues:

1. There does not seem to be a way to persist the boot order. When a new UEFI boot entry is created from the provisioning process, it pre-empts the PXE booting. I would prefer to have some option to force net-boot first. **Is there a way to force net-booting priority?**
2. When I manually set PXE as the priority once the image is provisioned, the PXE boot fails (as expected, since the provisioner is not re-deploying the image), but then I get dropped into the UEFI configuration screen instead of the second boot device being tried.

I want the failure of pxe booting to result in the next non-network device to boot instead of it to fall back into the UEFI firmware settings menu.

Is this expected behaviour?

[image]

[image]

TheMindVirus, 2023-02-25:
Apparently iPXE scripting lets you use `goto` statements now for jumping to previous labels: https://ipxe.org/scripting
but it is an iPXE issue rather than Pi4 UEFI. I have found the boot order to only persist with specific versions of RPI_EFI.fd and not with others, as there was an issue with saving boot variables on USB disks, then with SD cards, then with USB disks again.
Network boot became the priority either way (HTTP).

Banshee1221, 2023-08-12:
I've changed my approach and am using MAAS for now. I'm still encountering this issue.

Whenever MAAS deploys the image to the Pi, a new `ubuntu` entry is written and sent to the top of the boot order. This appears to happen with 1.31, 1.33 and 1.34. The Pi is configured with Boot Discovery Policy as `minimal`

UEFI running off of SD card. I have confirmed that the settings persist across reboots.

@TheMindVirus do you know which version of the `RPI_EFI.fd` could work?

TheMindVirus, 2023-08-12:
The versions I based my display mods from were 1.28, 1.29 and 1.33, 1.33 seems to be in common.
In each one, especially after boot order fixes, it sometimes started with HTTP boot.
https://github.com/TheMindVirus/pftf-rpi4

Banshee1221, 2023-08-13:
I am not having luck. I guess I'll have to dig into the code sometime. 

I've tried it again:

## 1.31 through 1.34
- copy to sd
- boot sd and enter menu
  - disable 3gb limit
  - set pxe to first
  - boot policy minimal
- cold reboot Pi
- confirm that settings are applied
- cold power off Pi
- set MAAS to provision, turn on Pi.
- MAAS completes, triggers reboot
  - automatic reboot is all good. No new entry. Things are working with pxe as default

If I reboot the Pi, warm or cold, a new `ubuntu` entry is present at the top of the boot order. Making PXE non-default. This means that there will always have to be physical intervention to reprovision.

I haven't tried 1.35 because I get boot errors with it.

---
**Note**: I **seem** to have been able to work around it by flashing the SD card, booting for the first time and changing the config, shutting down the Pi and setting the SD card to write-protected using a Linux box and this code: https://github.com/BertoldVdb/sdtool (but I also set it on a Windows box with disktool)

With the SD as readonly, I've been able to provision repeatedly with PXE staying at the top and no new `ubuntu` entry being added. Not an ideal solution.

3DJupp, 2024-02-15:
Hey, I am experiencing the same. Would love to have a generic Image for all Pis, no matter if they are 4 or 8GB and:
- Remove the memory limit to 3gb
- set UEFI PXE as default boot option. This does only work on the Pi where i created the files, like RPI_EFI.fd etc.