pftf/RPi4 #232: rpi4-uefi.dev not working

lexfrei, 2023-03-10:
I can't reach the site.

```
a.sviridkin@192:~$ curl rpi4-uefi.dev
curl: (6) Could not resolve host: rpi4-uefi.dev
✗: 6 @ Sat Mar 11 00:19:38 MSK 2023
```

The domain is still registered, so looks like just a DNS issue

```
a.sviridkin@192:~$ dig rpi4-uefi.dev @8.8.8.8

; <<>> DiG 9.10.6 <<>> rpi4-uefi.dev @8.8.8.8
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 51948
;; flags: qr rd ra ad; QUERY: 1, ANSWER: 0, AUTHORITY: 1, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;rpi4-uefi.dev.			IN	A

;; AUTHORITY SECTION:
rpi4-uefi.dev.		234	IN	SOA	ns-cloud-b1.googledomains.com. cloud-dns-hostmaster.google.com. 3 21600 3600 259200 300

;; Query time: 44 msec
;; SERVER: 8.8.8.8#53(8.8.8.8)
;; WHEN: Sat Mar 11 00:17:34 MSK 2023
;; MSG SIZE  rcvd: 135

✓ @ Sat Mar 11 00:17:34 MSK 2023
```

xgpt, 2023-08-11:
Still not fixed.

ringods, 2023-10-16:
@pbatard do you have access to get the website up and running again?

pbatard, 2023-10-16:
I don't.

ringods, 2023-10-16:
@pbatard the logical follow up question on my end is then: do you know who has this access and can you loop them in on this issue?

pbatard, 2023-10-16:
I believe @andreiw has been pinged multiple time on this issue.

ringods, 2023-11-18:
In the meantime: https://web.archive.org/web/20230103162918/https://rpi4-uefi.dev/

stuyod01, 2024-01-09:
I can provide the zip archive of Andrei's wordpress site if anyone would like to resurrect that.

Another option would be to harvest the content and create markdown or wiki pages on github.

ringods, 2024-01-10:
@stuyod01 do provide the zip file as an attachment to a comment here. I will have a look if I can make a markdown powered static site.

pbatard, 2024-01-10:
For the record, if someone is willing to make a [GitHub static web site](https://pages.github.com/), I'll be happy to create a `gh-pages` branch of this repository to publish it, since it makes sense that the web site should be handled by this project and we can easily have rpi4-uefi.dev pointing to this content by [setting a `CNAME` file](https://docs.github.com/en/pages/configuring-a-custom-domain-for-your-github-pages-site/managing-a-custom-domain-for-your-github-pages-site).

ringods, 2024-01-10:
@pbatard Is this `pftp` organization linked to a company? If not, can we open it up to more user contributions? For example, I would like to reach out to the author of the RPi5 initiative (see comment link) to bundle the efforts under this organization and have that repo moved to here as `RPi5`, similar to the existing `RPi3` and `RPi4`.

https://github.com/pftf/RPi4/issues/243#issuecomment-1877351217

Can you create a separate repository for the website? The website is not RPi v4 specific and I would like to see RPi 3/4/5 info ending up there. In the meantime, I registered domain `rpi-uefi.dev` again and paid for it myself. Consider this my first contribution to this project. 😄

pbatard, 2024-01-10:
> Is this pftp organization linked to a company?

Not really. It's a loose association of individuals who share(d) a common interest in making the Raspberry Pi platform officially supported by [EDK2](https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi) (along with other things such as trying to promote [SBSA/SBBR](https://en.wikipedia.org/wiki/Server_Base_System_Architecture)). As long as you share that goal and are willing to contribute, then the doors of pftf (which stands for _Pi Firmware Task Force_) are opened to you.

> I would like to reach out to the author of the RPi5 initiative (see comment link) to bundle the efforts under this organization

Please note that, at this stage, and to keep in line with the goal stated above, I feel like we should only accept an RPi5 project under pftf once its codebase has been submitted and integrated into EDK2 (which I am assuming/hoping is what Mario is working towards, but I am not completely sure).

On the other hand, I don't have a problem creating a  __separate__ `RPi5-dev` repo here, or something else, if we feel that's what needed for EDK2 integration, but I cannot stress enough that, from where I stand, __formal__ integration into EDK2 needs to happen, just as it happened for the Pi 3 and the Pi 4, before we should consider publishing downloadable UEFI firmware packages under pftf. And yes, I understand that this might be seen as an unnecessary hurdle/annoyance to many, but IMO, the benefits of doing it that way far outweigh the costs, as it'll mean that we will work with a codebase that has been properly vetted and reviewed, and also, it'll mean that people from all horizons will be able to see the Pi5 platform as something that can be officially relied on for UEFI, as opposed to something that a handful of individuals worked on at one stage, and that might cease to be maintained from one day to the next. Considering the interest that exists with regards to UEFI on the Pi 5, I believe that there should be enough individuals besides Mario wanting to contribute to the effort of having it officially integrated as an EDK2 platform (because by all means, Mario should not have to do it alone) so I don't see why we shouldn't be able to accomplish with the Pi 5 what was accomplished with the Pi 3 and Pi 4.

> Can you create a separate repository for the website?

I'd rather we restore the RPi4 specific `rpi4-uefi.dev` website first, and see where we go from here. I don't have much of an objection to having a more generic `rpi-uefi.dev` hosted under pftf, but the immediate goal here is to have the `rpi4-uefi.dev` website work again, since there are quite a few URLs that link to it, and I don't see how trying to go in too many directions at once will help with this goal.

So can I ask that we please do things one step at a time and complete the restoration of `rpi4-uefi.dev` first?

ringods, 2024-01-10:
> So can I ask that we please do things one step at a time and complete the restoration of `rpi4-uefi.dev` first?

Sure. Thas was my intent all along. However, I made a mistake in checking the domain name to have a typo and forgot the `4` in it. Noticing that the domain (with my typo) was available, I claimed it. Silly me. 🙈

samerhaj, 2024-01-10:
I already registered and own **rpi4-uefi.dev**

I can gladly point it to the new Github static web site when ready

stuyod01, 2024-01-10:
I could not attach it due to the 25MB limit.  Here is a link to the file on my google drive: https://drive.google.com/file/d/1J3_wVqVgmlTtlu0IKSgdBK_WbyDKp_FK/view?usp=sharing

samerhaj, 2024-02-10:
I temporarily redirected https://rpi4-uefi.dev/ to https://github.com/pftf/ . I can edit the URL to whatever documentation page we end up with on the PFTF Github

closed 2025-05-25 by ardbiesheuvel