pftf/RPi4 #236: Setup UI bugs out over serial connection

jdoss, 2023-06-20:
I am trying to use this on a Turing Pi 2 board with CM4s and when I connect to a CM4 over serial with `microcom -s 115200 /dev/ttyS2` the Setup UI slowly bugs out as I navigate through the menus until it becomes totally unusable and locks up.  

[image]

I have also tried `picocom -b 115200 --nolock /dev/ttyS2` which is even worse when using the Setup UI. 
[image]

Are there any kind of settings I should try when connecting over serial to fix this issue when accessing the Setup UI?

pbatard, 2023-06-29:
What other serial communication utilities have you tried besides microcom?

If you only see the issue in microcom, then you should ask in the microcom user support forums, as you are unlikely to find people that familiar with this utility here...

jdoss, 2023-06-29:
@pbatard I get the same results with picocom.