pftf/RPi4 #238: Can you Make It So theres no test mode text and add wi-fi

Bryce10111, 2023-06-26:


pbatard, 2023-06-29:
WiFi support is unlikely to happen, as it would require a lot of work, and regular contributors to this project, who are also participating in other endeavours (Open Source or otherwise) and therefore have to carefully select what new developments they take on, are unlikely to allocate time anytime soon to add support for the WiFi hardware found on the Pi 4.

As to removing text mode, first of all, it's hard to imagine why this should be considered as something that is needed (unless you describe in which manner it negatively impacts users) and, at any rate, the boot output comes from what the [EDK2 project](https://github.com/tianocore/edk2) defaults too, so you would have to submit a patch to EDK2 and get it accepted there, before you would be able to get what you request here. But then again, you would have to make a strong case to the EDK2 project as to why you want to add such a feature.

Since I believe I answered your questions, I will close this issue.

closed 2023-06-29 by pbatard

Bryce10111, 2023-07-01:
But Can You Make It so there is boot key and its f11 and it shows a window it shows all the boot devices

Bryce10111, 2023-07-01:
I Can Make Requests For Updates For The UEFI If You Want Me too

Bryce10111, 2023-07-01:
Like Can You Make It so It Shows the windows logo after it shows the raspberry pi logo

Bryce10111, 2023-07-01:
Can You Make Minecraft Work