pftf/RPi4 #240: UEFI firmware installation fails: Could not detect network connection

nandlab, 2023-08-17:
I have prepared a FAT32 USB stick with the UEFI firmware files.

It boots fine and shows a colored Pi logo and a progress bar. When the progress nearly reaches the end, it stops and shows the following error messages on the screen:

```
  Error: Could not detect network connection.

  Error: Could not detect network connection.
BdsDxe: No bootable option or device was found.
BdsDxe: Press any key to enter the Boot Manager Menu.
```
When I press a key I see a configuration GUI.

What does it mean? Is network connection required for the UEFI firmware installation?

pbatard, 2023-08-17:
The firmware is just that: A firmware.

It's like when you buy a modern PC with no operating system installed. The UEFI firmware will boot and, if it can't find any OS installed, it may _fallback_ to trying to boot from the network through PXE, in case you have a PXE image server.

The RPI4 UEFI firmware behaves the same. It expects that __you__ will have taken the steps to install an OS, or provide the botable files required for OS installation. But it does not provide the OS for you. Just like when you buy a bare PC, it does not provide the OS for you, just the UEFI components that allow you to install the OS.

For instance, if you wanted to install Linux Debian in UEFI mode using the UEFI firmware, you could follow [this guide](https://forums.raspberrypi.com//viewtopic.php?f=50&t=282839), which, you will see, requires providing the OS installation files alongside the __bare__ UEFI firmware.

closed 2023-08-17 by pbatard

nandlab, 2023-08-18:
Thank you for the explanation!