pftf/RPi4 #241: Secure Boot Setup

nandlab, 2023-08-18:
How would you configure the RPi 4 so that only a trusted OS can be booted (full chain of trust from the hardware to the OS kernel)?

As I understand, the UEFI firmware supports secure boot.

Should the UEFI firmware itself also be signed and verified by the SoC firmware to prevent an attacker from using a different booting mechanism (using something like [this](https://github.com/raspberrypi/usbboot/tree/master#secure-boot))?

jlinton, 2023-12-21:
It does, but it is not secure since anything that gains root privileges in the OS, can simply replace the secure boot keys enrolled on the machine. Making it largely worthless without a HW enforced root of trust and chain.

nandlab, 2023-12-22:
I managed to set up secure boot using the Raspberry Pi's own [method](https://github.com/raspberrypi/usbboot/tree/master/secure-boot-example). So having secure boot on the Raspberry Pi does not require UEFI.
But having UEFI with hardware enforced secure boot would likely be possible if you create a boot.img FAT image with the UEFI firmware files and a boot.sig file with your digital signature.

closed 2023-12-22 by nandlab