pftf/RPi4 #242: FreeBSD boot stuck at armv8crypto0: CPU lacks AES instructions

ykla, 2023-09-21:
```

Consoles: EFI console  
    Reading loader env vars from /efi/freebsd/loader.env
Setting currdev to disk0p1:
FreeBSD/arm64 EFI loader, Revision 1.1

   Command line arguments: loader.efi
   Image base: 0x3378e000
   EFI version: 2.70
   EFI Firmware: https://github.com/pftf/RPi4 (rev 1.00)
   Console: efi (0x1000)
   Load Path: \EFI\BOOT\BOOTAA64.EFI
   Load Device: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/USB(0x1,0x0)/HD(1,GPT,A3
D6A4C4-53AA-11EE-A7BA-002590EC5BF2,0x3,0x10418)
   BootCurrent: 0001
   BootOrder: 0000 0001[*] 0002 0003 0004 0005 0006 0007
   BootInfo Path: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/USB(0x1,0x0)
Ignoring Boot0001: Only one DP found
Trying ESP: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/USB(0x1,0x0)/HD(1,GPT,A3D6A4
C4-53AA-11EE-A7BA-002590EC5BF2,0x3,0x10418)
Setting currdev to disk0p1:
Trying: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/USB(0x1,0x0)/HD(2,GPT,A3D6A4CC-5
3AA-11EE-A7BA-002590EC5BF2,0x1041B,0x21ED00)
Setting currdev to disk0p2:
Loading /boot/defaults/loader.conf
Loading /boot/defaults/loader.conf
Loading /boot/device.hints
Loading /boot/loader.conf
Loading /boot/loader.conf.local
?c\
   ______               ____   _____ _____  
  |  ____|             |  _ \ / ____|  __ \ 
  | |___ _ __ ___  ___ | |_) | (___ | |  | |
  |  ___| '__/ _ \/ _ \|  _ < \___ \| |  | |
  | |   | | |  __/  __/| |_) |____) | |__| |
  | |   | | |    |    ||     |      |      |
  |_|   |_|  \___|\___||____/|_____/|_____/      ```                        `
                                                s` `.....---.......--.```   -/
 /---------- Welcome to FreeBSD -----------\    +o   .--`         /y:`      +.
 |                                         |     yo`:.            :o      `+-
 |  1. Boot Installer [Enter]              |      y/               -/`   -o/
 |  2. Boot Single user                    |     .-                  ::/sy+:.
 |  3. Escape to loader prompt             |     /                     `--  /
 |  4. Reboot                              |    `:                          :`
 |  5. Cons: Serial                        |    `:                          :`
 |                                         |     /                          /
 |  Options:                               |     .-                        -.
 |  6. Kernel: default/kernel (1 of 1)     |      --                      -.
 |  7. Boot Options                        |       `:`                  `:`
 |                                         |         .--             `--.
 |                                         |            .---.....----.
 \-----------------------------------------/
   Autoboot in 8 seconds. [Space] to pause  

Loading kernel...
/boot/kernel/kernel text=0x2a8 text=0x9b1d90 text=0x25c134 data=0x14cb38 data=0x
0+0x2c5000 0x8+0x14b770+0x8+0x173d94
Loading configured modules...
can't find '/boot/entropy'
can't find '/etc/hostid'
Using DTB provided by EFI at 0x35a21000.
EFI framebuffer information:
addr, size     0x3e91b000, 0x1d4c00
dimensions     800 x 600
stride         800
masks          0x00ff0000, 0x0000ff00, 0x000000ff, 0xff000000
---<<BOOT>>---
WARNING: Cannot find freebsd,dts-version property, cannot check DTB compliance
Copyright (c) 1992-2023 The FreeBSD Project.
[... 248 lines cut ...]
cpu0: <Open Firmware CPU> on cpulist0
bcm2835_cpufreq0: <CPU Frequency Control> on cpu0
pcib0: <BCM2838-compatible PCI-express controller> mem 0x7d500000-0x7d50930f irq 80,81 on simplebus2
pcib0: hardware identifies as revision 0x304.
pci0: <OFW PCI bus> on pcib0
pcib1: <PCI-PCI bridge> irq 91 at device 0.0 on pci0
pci1: <PCI bus> on pcib1
bcm_xhci0: <VL805 USB 3.0 controller (on the Raspberry Pi 4b)> irq 92 at device 0.0 on pci1
bcm_xhci0: 32 bytes context size, 64-bit DMA
usbus2 on bcm_xhci0
genet0: <RPi4 Gigabit Ethernet> mem 0x7d580000-0x7d58ffff irq 82,83 on simplebus2
genet0: GENET version 5.0 phy 0x0000
miibus0: <MII bus> on genet0
brgphy0: <BCM54213PE 1000BASE-T media interface> PHY 1 on miibus0
brgphy0:  10baseT, 10baseT-FDX, 100baseTX, 100baseTX-FDX, 1000baseT, 1000baseT-master, 1000baseT-FDX, 1000baseT-FDX-master, auto
genet0: Ethernet address: d8:3a:dd:27:a8:b3
clk_fixed4: <Fixed clock> disabled on ofwbus0
clk_fixed4: Cannot FDT parameters.
device_attach: clk_fixed4 attach returned 6
clk_fixed4: <Fixed clock> disabled on ofwbus0
clk_fixed4: Cannot FDT parameters.
device_attach: clk_fixed4 attach returned 6
gpioled0: <GPIO LEDs> on ofwbus0
armv8crypto0: CPU lacks AES instructions
```

void-fm, 2024-01-19:
I can confirm this still happens with latest v 1.35

markmi, 2024-01-19:
"armv8crypto0: CPU lacks AES instructions" is not a fatal error
but just informational. Here is part of a "dmesg -a" from  an
RPi3B boot using the normal U-Boot style:

gpioled0: <GPIO LEDs> on ofwbus0
gpioled0: <ACT> failed to map pin
armv8crypto0: CPU lacks AES instructions
Timecounters tick every 1.000 msec
usbus1: 480Mbps High Speed USB v2.0
ugen1.1: <DWCOTG OTG Root HUB> at usbus1
uhub0 on usbus1

The message is not special to RPi4B's or to EDK2 use at all.
Nor does what it reports lead to stopping the boot sequence.

The actual problem is a FreeBSD kernel mishandling of reserving
memory: a rejection/panic of a valid ACPI set up (valid as far
as I can tell).

wb7odyfred, 2024-05-09:
I am new, this is a change suggested to me, that allowed FreeBSD 15.0-CURRENT and GhostBSD-Arm64 14.0-STABLE  to boot up.  RPI 5 A76 cpu core with Armv9  might support some of these kernel modules from FreeBSD.org  aarch64 software.
[loader.conf.txt](https://github.com/pftf/RPi4/files/15265856/loader.conf.txt)
[rc.cont.txt](https://github.com/pftf/RPi4/files/15265857/rc.cont.txt)

edit  /boot/loader.conf file   disable loading of aesni , geom_eli , and  crypto modules.
[Random thoughts and disable loadable modules ](https://ghostbsd-arm64.blogspot.com/2023/12/errors-nginx-dhcp-boot-ntpd-ntpdate.html)

RPI 4 A72 cpu core does not support aesni instruction.  RPI 5 A76 cpu core might support armv9 , aesni cpu instruction 
note: aesni, geom_eli.ko, crypto  These three x86-64 kernel modules should not be loaded for aarch64 , arm64 kernel.  Just comment them out for now, or change "YES" to "NO" for aesni_load="NO".

or  please don't edit /boot/defaults/loader.conf though, that should not be necessary.  So do edit /boot/loader.conf file

 # Load geom support
 crypto_load="NO"
 geom_eli_load="NO"  #maybe YES in this option

# Enable hardware crypto
 aesni_load="NO"

attached file names have  add '.txt' suffix  for  /boot/loader.conf  and /etc/rc.conf file   From the files look like I just needed to set crypto_load="NO"    Welcome your comments as what works for you in booting FreeBSD aarch64 on your Arm64 hardware, for the other 2 modules  geom_eli_load="NO" ; aesni_load="NO"  ; .  Fred

ykla, 2024-05-10:
> I am new, this is a change suggested to me, that allowed FreeBSD 15.0-CURRENT and GhostBSD-Arm64 14.0-STABLE  to boot up.  RPI 5 A76 cpu core with Armv9  might support some of these kernel modules [...]

It is actually possible to boot FreeBSD 14 and 15 (using the rpi4 image) on a Raspberry Pi 5 without making any changes. Just use https://github.com/worproject/rpi5-uefi. I haven't tested Raspberry Pi 4 though.