pftf/RPi4 #243: Meta: Plans for RPi5?

sundhaug92, 2023-10-01:
Hi
Are there any plans for RPi5 support?

markjsones, 2023-11-04:
Seconded, I desperately want to have this same functionality on Pi5

supersjimmie, 2023-11-07:
Just received my Raspberry Pi 5 (8GB), almost ready to install the VMware ESXi Fling.
All I need now is the rpi5 uefi bit...

sundhaug92, 2023-11-08:
> Just received my Raspberry Pi 5 (8GB), almost ready to install the VMware ESXi Fling. All I need now is the rpi5 uefi bit...

AFAIK VMware ESXi Flink does not yet support the RPi 5, and UEFI alone won't fix that

supersjimmie, 2023-11-08:
> > Just received my Raspberry Pi 5 (8GB), almost ready to install the VMware ESXi Fling. All I need now is the rpi5 uefi bit...
> 
> AFAIK VMware ESXi Flink does not yet support the RPi 5, and UEFI alone won't fix that

Sure, agree. However without uefi it won't be possible to make ESXi compatible.
So we need uefi first to then make ESXi work.

tester962, 2023-11-08:
It would be better to ask Raspberry Pi about it. The hardware vendor should be the one providing support for this (just like in the x86 world...), particularly since from what I've heard they haven't even bothered donating some boards to the cause.

The hardware itself is still broken and far away from anything resembling SystemReady. As such, UEFI alone is not too useful. The board needs quite a bit of driver work on all major OSes, and it's highly unrealistic to expect any progress on that front, when even for RPi 4 things aren't great and all development/interest pretty much came to a halt over a year ago.

For ESXi: there is no Ethernet driver and the SDK is behind a paywall. You'd be stuck with USB Ethernet.

esloof, 2023-11-11:
[image]

I've tried the [Raspberry Pi 4 UEFI Firmware Image](https://github.com/pftf/RPi4#raspberry-pi-4-uefi-firmware-images) after changing the config.txt and adding os_check=o the screen stays back.

supersjimmie, 2023-11-13:
Same result @esloof , that was my quick attempt too.
After that first attempt I updated the rpi bios, but also no result.
I guess it's not THAT easy.

sundhaug92, 2023-11-24:
Yeah you need more than just the dtb file, you need someone to write the appropriate drivers etc upstream. There seems to be movement in it, given that [Someone already has W11 runninng](https://twitter.com/mariobalanica/status/1728121929691717776)

supersjimmie, 2023-11-28:
> Yeah you need more than just the dtb file, you need someone to write the appropriate drivers etc upstream. There seems to be movement in it, given that [Someone already has W11 runninng](https://twitter.com/mariobalanica/status/1728121929691717776)

I start to believe this is fake.
There's really nothing else to find about UEFI for the rpi 5.
Not even a reference on his own github https://github.com/mariobalanica

Marcinoo97, 2023-12-01:
> I start to believe this is fake. There's really nothing else to find about UEFI for the rpi 5. Not even a reference on his own github https://github.com/mariobalanica

its unreleased because it lacks display output support.  screenshot was via RDP.

sundhaug92, 2023-12-02:
> > > > Yeah you need more than just the dtb file, you need someone to write the appropriate drivers etc upstream. There seems to be movement in it, given that [Someone already has W11 runninng](https://twitter.com/mariobalanica/status/1728121929691717776)
> > > 
> > > 
> > > I start to believe this is fake. There's really nothing else to find about UEFI for the rpi 5. Not even a reference on his own github https://github.com/mariobalanica
> > 
> > 
> > its unreleased because it lacks display output support. screenshot was via RDP.
> 
> So he DOES have UEFI working? Because that's what we're looking for here.

He only has it partially working

supersjimmie, 2023-12-04:
> He only has it partially working

But isn't that one of the purposes of sharing on github? To work on things that partially work, so other people can be involved?

Marcinoo97, 2023-12-04:
> But isn't that one of the purposes of sharing on github? To work on things that partially work, so other people can be involved?

so remove free will from people and force all software devs to open source anything they ever worked on?

supersjimmie, 2023-12-05:
> so remove free will from people and force all software devs to open source anything they ever worked on?

Sorry for creating misunderstanding.

That's not what I am saying.
However the ARM UEFI however is open source, isn't it?
What I ment was just the other way around, he could try to do all on his own or he could share, no force, just a suggestion. github forces nobody, I force nobody.

Marcinoo97, 2023-12-05:
display is working now.  but release is not happening soon as code isn't formatted in a way edk2 mailing list would approve.

supersjimmie, 2023-12-05:
And sorry @Marcinoo97 that I missed the WoR project community outside of github.

Marcinoo97, 2023-12-11:
possibly something soon.  https://github.com/worproject/edk2-platforms

jlinton, 2023-12-21:
I have a pi5 as well, but I also need to upstream a pile of pi4 patches before I get serious about working on the pi5.

Marcinoo97, 2023-12-21:
but also wait for mario to drop some real support for rpi5. he still haven't uploaded any acpi tables tho.

petrjac, 2024-01-04:
https://github.com/worproject/rpi5-uefi

3DJupp, 2024-02-08:
> AFAIK VMware ESXi Flink does not yet support the RPi 5, and UEFI alone won't fix that

I would love to see esxi (or other customized uefi installs) using preconfigured !=3GB RAM UEFI FIles.
There is no known way to customize the uefi.zip, or not for me. 🤔
Currently, i do use a customized 8gb file for both 4 and 8 GB boards, which seems to work, 
**example ks.cfg file** for stateless installs (only the %post section, which utilizes mcopy etc. 
Shoutout to this amazing blog.: https://williamlam.com/2020/11/stateless-esxi-arm-with-raspberry-pi.html (I adopted this, so it fits my needs)
```
%post --interpreter=busybox
mkdir -p /tmp/uefi

# To-Do: Select UEFI File by RAM-Size
wget http://192.168.178.100/uefi_pi4_8gb_1.35.zip -O /tmp/uefi/uefi.zip
unzip /tmp/uefi/uefi.zip -d /tmp/uefi/
rm -f /tmp/uefi/uefi.zip
# Get USB-Device from weasel.log
USB_DEVICE=$(grep "install --firstdisk ==" /var/log/weasel.log | grep -v grep | awk '{print $6}')

# Copy uefi.zip to USB Drive
mcopy -i "/dev/disks/${USB_DEVICE}:1" -vs /tmp/uefi/* ::/
```

nomi-ramzan, 2024-05-29:
Any update on this ?

kfox1111, 2025-04-28:
Same question. There is https://github.com/worproject/rpi5-uefi, but has gone archived. It does work, so would be a shame if its lost.

heini, 2025-08-03:
> Same question. There is https://github.com/worproject/rpi5-uefi, but has gone archived. It does work, so would be a shame if its lost.

Maybe PFTF can talk to @mariobalanica and fork the repository under their umbrella. In https://github.com/worproject/rpi5-uefi/issues/18 he has clearly expressed that he wont put any further work into it...

jlinton, 2025-08-05:
I tend to agree with most of what [pbatard] and [mariobalanica] here mentioned. This is the 5th major rpi release, and each time a group of people stepped up in the community to assist with upstreaming and getting the platform working with a wider range of OS's than just Raspberry Pi OS. And each time they were working around hardware engineering decisions that frankly, don't make any sense for a general purpose device. Ex: non-coherent PCIe, which kills perf by creating software bound cache management, or PCIe devices that have all the functions in a shared address space rather than broken out into individual functions as the standard dictates. Never mind the borderline ones like trying to support cert timestamp validations in the firmware+OS without a RTC, or filling the limited SPI flash with functionality that is standardized higher in the firmware/OS stack.

But frankly, in a lot of cases those weren't what pushes people away, its the fact that the hardware vendors won't do the minimum required to support the community efforts. Be that regression testing new versions of their proprietary firmware against the upstream efforts, or taking bug reports seriously. I just saw that apparently the pi4 XHCI mailbox reset bug I reported a few years ago, was finally fixed when someone discovered it could be duplicated with RPiOS. The lack of documentation is just the first sign, and while its frequently possible to reverse engineer just about anything (see Asahi Linux!) it starts to feel disrespectful when a company benefits from those efforts, which take 10x, vs just having the documentation, then changes something in the platform that results in further long hours of community work to reach parity with the previous version. 

So, from the past generation, to the next, take this as a warning. If you don't get the parent hardware vendor to buy into providing more than lip service support, its likely that when you approach them with bugs, suggestions, and recommendations for the next generation, they will ignore you. And you will soon question why your putting in extra effort for little return when the next generation has problems they were warned about.

PS: If that wasn't clear: The PI has basically all the firmware abstractions one would expect out of an ACPI like platform, but implemented in a completely proprietary manner that isn't quite a match. This means your getting all the unfixable bugs of an x86 ACPI platform without the standardization that makes it 95% work out of the box. There is frankly little excuse for this, given we are now seeing vendors that understand a little extra work upfront saves them a lot of pain longer term.