pftf/RPi4 #250: RPI_EFI.fd file

phanbagiabao2001, 2024-03-04:
In the [RPi4_UEFI_Firmware_v1.35.zip](https://github.com/pftf/RPi4/releases/download/v1.35/RPi4_UEFI_Firmware_v1.35.zip) from https://github.com/pftf/RPi4/releases there are a file RPI_EFI.fd (generated firmware). 
Can anyone explain me what does this file contain and how to extract this file to know its components?
I found a link that helps build it, https://developer.arm.com/documentation/102677/0200/Set-up-the-Raspberry-Pi-4/Install-UEFI

pbatard, 2024-03-04:
> what does this file contain

It contains exactly the components that are described in the fdf: https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/RPi4/RPi4.fdf

If you need more than this, or how to interpret the content of the fdf, you will have to look for the EDK2/UEFI documentation on the matter.

> and how to extract this file to know its components?

UEFITool should be what you are looking for: https://github.com/LongSoft/UEFITool

closed 2024-03-04 by pbatard

pbatard, 2024-03-06:
Upstream EDK2 __is__ pretty much this repo. This repo is just a build script that fetches the upstream EDK2 source to build the Pi 4 firmware, per https://github.com/pftf/RPi4/blob/master/.github/workflows/linux_edk2.yml.

There are no updates to the firmware because (AFAIK) there have not been updates to upstream EDK2 code to warrant one (see [here](https://github.com/tianocore/edk2-platforms/commits/master/Platform/RaspberryPi) -- the last 2 updates since last year are just run of the mill EDK2 that all platforms got and that don't warrant a new release). When there are upstream updates to the EDK2 code, I (or somebody else) will update the repo to point to latest EDK2 and produce a new release.

phanbagiabao2001, 2024-03-07:
> It contains exactly the components that are described in the fdf: https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/RPi4/RPi4.fdf
>
> If you need more than this, or how to [...]
Thanks for your answers. Can I build the firmware based on edk2 for other platforms which are not in list https://github.com/tianocore/edk2-platforms/tree/master/Platform ?

pbatard, 2024-03-07:
> can I build the .fd file based on edk2 for other platforms which are not in list

That's very unlikely.

For instance, if all it took to get a platform supported was to pick existing edk2-platforms components and shove them into a new `.fdf`, we would long have added UEFI support for the Raspberry Pi 5, since it's a platform that's supposed to be close to the Pi 4.

Unfortunately, it doesn't work that way. The components referenced in the `.fdf` must point to code that is designed __specifically__ for the platform, and, if you don't have the underlying code for a specific hardware component, then you will need to write it from scratch before you can reference it in a `.fdf`. And as you can imagine, if the Pi 4 and Pi 5 are different enough that it needs new code, it's very very unlikely that you will be able to get away with support for whatever platform you have in mind simply by crafting a new `.fdf`.

Considering that everything in edk2-platforms is public, I would strongly urge you to familiarize yourself with what goes into a platform before asking further question. If you are interested in developing for your own platform, there is no shortcut to picking up an existing platform apart, to understand how it all works, before you start working on your own.