pftf/RPi4 #252: Display not working with v1.36

alsldlflglhljlk, 2024-03-15:
I am running Arch Linux ARM on my Raspberry Pi 4 (yes I know there is a specific Raspberry Pi version, but I am simply testing for compatibility purposes). I was able to sucessfully boot from it with v1.35, but I have lost my display with v1.36. However, this display issue only occurs sometime during the systemd boot process, as by invoking `fs0:\Image root=/dev/sda2 init=/bin/bash rootwait ro` at the UEFI shell, I am able to enter a rescue shell. I am still able to SSH into the machine, so I can still try to debug it.

dmesg:
```[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.7.9-1-aarch64-ARCH (builduser@leming) (aarch64-unknown-linux-gnu-gcc (GCC) 12.1.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT_DYNAMIC Tue Mar 12 11:20:09 MDT 2024
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: EFI v2.7 by https://github.com/pftf/RPi4
[    0.000000] efi: ACPI 2.0=0x33a20018 SMBIOS=0x37100000 SMBIOS 3.0=0x370e0000 MEMATTR=0x35200018 INITRD=0x33881998 RNG=0x33a2f918 MEMRESERVE=0x33881818 
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000000200000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000000200000..0x00000000041fffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003ee642c0..0x000000003ee646bf (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ee64700..0x000000003ee6474c (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      em 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    em 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: em 0x0000000000000000-0x00000000001fffff]
[    0.000000]   node   0: em 0x0000000000200000-0x000000003381ffff]
[    0.000000]   node   0: em 0x0000000033820000-0x000000003387ffff]
[    0.000000]   node   0: em 0x0000000033880000-0x0000000033881fff]
[    0.000000]   node   0: em 0x0000000033882000-0x0000000033a1ffff]
[    0.000000]   node   0: em 0x0000000033a20000-0x0000000036efffff]
[    0.000000]   node   0: em 0x0000000036f00000-0x00000000372dffff]
[    0.000000]   node   0: em 0x00000000372e0000-0x000000003b2fffff]
[    0.000000]   node   0: em 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 em 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19712 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 22 pages/cpu s51624 r8192 d30296 u90112
[    0.000000] pcpu-alloc: s51624 r8192 d30296 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: BOOT_IMAGE=/Image root=/dev/sda2 rootwait ro
[    0.000000] Unknown kernel command line parameters "BOOT_IMAGE=/Image", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped em 0x0000000037300000-0x000000003b300000] (64MB)
[    0.000000] Memory: 3777088K/4049920K available (19392K kernel code, 4576K rwdata, 10828K rodata, 8320K init, 972K bss, 207296K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Dynamic Preempt: full
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[... 499 lines cut ...]
[   14.284148] brcmfmac: brcmf_set_channel: set chanspec 0x100c fail, reason -52
[   14.284437] brcmfmac: brcmf_set_channel: set chanspec 0x100d fail, reason -52
[   14.284715] brcmfmac: brcmf_set_channel: set chanspec 0x100e fail, reason -52
[   14.285008] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[   14.388200] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[   14.492109] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[   14.596084] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[   14.700130] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[   14.700418] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[   14.700696] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[   18.399797] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   18.399822] Bluetooth: hci0: Failed to set baudrate
[   18.402064] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[   18.402716] Registered IR keymap rc-cec
[   18.402867] rc rc0: vc4-hdmi-1 as /devices/platform/soc/fef05700.hdmi/rc/rc0
[   18.402973] input: vc4-hdmi-1 as /devices/platform/soc/fef05700.hdmi/rc/rc0/input8
[   18.403270] vc4_hdmi fef05700.hdmi: Could not register PCM component: -517
[   18.418555] platform gpu: deferred probe pending
[   20.447801] Bluetooth: hci0: command 0x0c03 tx timeout
[   28.639798] Bluetooth: hci0: BCM: Reset failed (-110)
```

Attached is a low-quality video of how the display is lost upon systemd startup.

https://github.com/pftf/RPi4/assets/119443367/bb7a7e12-74f4-4b1f-af05-e6af187c3b4e

alsldlflglhljlk, 2024-04-09:
Alright, a new update (on Arch Linux) has rendered v1.35 without display as well. However, by adding `nomodeset` to the kernel command line, I can solve this issue. This indicates an issue with KMS not present in v1.34.

alsldlflglhljlk, 2024-04-10:
Maybe something [here](https://forums.raspberrypi.com/viewtopic.php?t=326393#p1953806)?

Roburetto, 2024-04-10:
"HDMI audio is inherently linked to the video mode, and running audio from the firmware and video from the kernel will run into issues should the video mode ever be changed or otherwise not match."

Makes sense. Same error message seems [to plague opensuse](https://lists.opensuse.org/archives/list/bugs@lists.opensuse.org/thread/Y4H3PZLZ4C3ASBZ6NPAFBBY4HCPJ3QEV/).

Have you tried Arch with pftf v1.34 ?

alsldlflglhljlk, 2024-04-10:
It works with v1.34 and below.

Roburetto, 2024-04-10:
opensuse devs came to the conclusion that changes in raspberrypi-firmware-dt from 2023/11/21 messed with the device tree. The solution being reverting and old version (as seemingly 1.34 and prior are including...?)

Ref.: https://bugzilla.opensuse.org/show_bug.cgi?id=1217512

"I am afraid that vendor changes around new RPi5 makes new
device trees unusable without corresponding changes in DMA
driver, which I don't see to happen soon."
(...)
I think the safest resolution for the moment is 
to revert to prior device tree version."

alsldlflglhljlk, 2024-04-10:
I copied the `.dtb` files from v1.34 and replaced the ones from v1.36, and... success! I'll file a PR for this soon.

closed 2024-04-10 by alsldlflglhljlk

pbatard, 2024-04-11:
Please try with the [newly released v1.37](https://github.com/pftf/RPi4/releases/tag/v1.37) that reverts the DTB to a version before the HDMI Audio changes and see if that works better.

Roburetto, 2024-04-12:
I can confirm that release 1.37 is working fine in a dual boot with Windows 11 & Fedora 40.

Roburetto, 2024-04-13:
Interesting discovery: booting current Raspberry Pi OS with dtb from pftf 1.37 mostly works but it is impossible to reboot or power down the device (the disk stays always on).

Booting current Raspberry Pi OS with DTB from pftf 1.36 works best.

I am assuming the kernel from Raspberry Pi OS expects a more recent DTB.

Roburetto, 2024-04-13:
Also, I am not sure mini-uart is working properly. I have to disable it to make bluetooth work both in Fedora and Raspberry Pi OS.

alsldlflglhljlk, 2024-04-15:
@Roburetto You should probably open a new issue containing applicable logs for the BT problem. I personally never got BT to work on any OS, getting  `Bluetooth: hci0: BCM: Reset failed (-110)` even with proper firmware.

jlinton, 2024-04-16:
The BT issues are almost assuredly because the Linux kernel cannot flip the secret console mux, and the config.txt is defaulting the firmware/linux consoles to the pl011. 

Although its a bit hit/miss the unmerged patch on the edk2 mailing lists to set the baud on both the pl011 and the miniuart fixes BT enough to pair with slow devices like keyboards in ACPI mode they still need to be inverted back to the defaults in config.txt because the BT/DT configs don't work in mainline otherwise.

Roburetto, 2024-04-17:
In my experience Bluetooth seems to work best if I switch both uart and mini-uart OFF in pftf's config.txt (the drawback being keyboard input on pftf becoming unstable).

On the Raspberry Pi OS disabling mini-uart alone renders the device unable to fully power off - something which is fixed when you additionally disable uart.

On Raspberry Pi OS I am now facing a different issue: Kwin busts in KDE (the desktop environment does start on X11 provided you disable enabling the composer on start-up but freezes the Pi on Wayland altogether).
It’s worth nothing that Wayland works fine with the default WM provided (LXDE over Wayfire I believe).