pftf/RPi4 #253: Programmatically change (default) settings

WizardUli, 2024-03-31:
Hello. I'm building sd card image for RPi (via Nix, but you can also imagine doing it using Buildroot etc..) and I'll like to patch this prebuilt UEFI firmware so it uses 4G RAM and devicetree so it can boot out of the box without any initial setup.
I understand such settings are stored directly inside the `RPI_EFI.fd`. What format are they stored in? Are they at deterministic offset? Is it feasible to write a script to change them?

valtzu, 2024-09-30:
You can patch it with `virt-fw-vars` command from `virt-firmware` python package (also available as `python3-virt-firmware` on Ubuntu). 

### Example:

https://github.com/valtzu/rpi-mkosi/blob/5b9f1eafdf0607041df6aed20cb2011cf26dea0f/mkosi.images/base/mkosi.prepare#L14-L21

https://github.com/valtzu/rpi-mkosi/blob/5b9f1eafdf0607041df6aed20cb2011cf26dea0f/efivars.json

WizardUli, 2024-09-30:
> You can patch it with `virt-fw-vars` command from `virt-firmware` python package (also available as `python3-virt-firmware` on Ubuntu).
> ### Example: [...]

Oooo. Thanks.

closed 2024-09-30 by WizardUli