pftf/RPi4 #255: Memory limit remain after disabling on v1.37

pieroproietti, 2024-04-15:
I installed the  last Raspberry Pi 4 UEFI Firmware **v1.37**, I deactivating the 3 GB limit of RAM, from the UEFI menu, this is saved, but the limit persist.

There are problems, someone can confirm? Thanks

[image]

pieroproietti, 2024-04-15:
This is the UEFI configuration, Limit to 3 GB Memory is disabled and saved.

[image]

alsldlflglhljlk, 2024-04-18:
`free -h` shows a total of `3.7 Gi`, indicating that the measurement is in gibibytes, not gigabytes. 3.7 gibibytes to gigabytes = 3.97284 gigabytes.

pieroproietti, 2024-04-20:
@alsldlflglhljlk, I have 8GB model, but yes You are right: this is not a problem of 3GB limit. I will check again.

closed 2024-04-20 by pieroproietti