pftf/RPi4 #262: Kernel loads extremely slowly on some SD cards

Gadget64, 2024-09-07:
Whenever I use the UEFI firmware on my [UHS-I SanDisk Extreme 64GB microSDXC](https://www.westerndigital.com/en-us/products/memory-cards/sandisk-extreme-uhs-i-microsd-160-mbps?sku=SDSQXA2-064G-AN6MA), it takes extremely long for the bootloader to load and start Linux. The process clearly gets stuck while loading the kernel / initrd into memory, as shown by `systemd-analyze`:

<pre>
Startup finished in 4.861s (firmware) + <b>1min 52.088s (loader)</b> + 3.983s (kernel) + 12.310s (userspace) = 2min 13.244s
</pre>

While I am using systemd-boot on my system, the exact same issue happens with GRUB. (Unfortunately, systemd-boot does not seem to have proper logging capabilities and I didn't manage to get useful info trying to debug this with GRUB yet.) Therefore, the only thing I can tell is that after the bootloader's picker is shown without issues and the boot has continued, it just stays at a black screen for a minute or two before Linux seems to have been loaded – which is when the following lines of text are shown at once:

<pre>
EFI stub: Booting Linux Kernel...
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...
</pre>

It seems that what took so long is for the kernel to load; and this is on my fresh install of NixOS, with a kernel image of just 70 MB. Once these messages have been shown after the long wait however, the Linux kernel starts booting quickly and properly as usual.

I can rule out that the card is not at fault, as it boots all non-UEFI configurations as quickly as expected – and even with UEFI, it performs very well once the system has gotten past the point of loading the kernel into memory. Additionally, an identical install of UEFI + NixOS starts the kernel almost immediately on an USB 2.0 flash drive, even though it is much slower than my microSD card.

Is the card maybe simply just running at a very low speed at this stage, before the Linux kernel properly configures it? It does indeed only report it as `sd high-speed` with `cat /sys/kernel/debug/mmc*/ios` (which should be 25 MB/s) before I've properly applied the device-trees, and only then is it shown as `sd uhs DDR50` (the expected UHS-I speed of 50 MB/s, which is what vanilla Raspberry Pi OS reports also), although both should be more than fast enough. Let me know if I can provide any more information!

pm4rcin, 2024-11-08:
I can confirm boot is really slow on Fedora IOT UEFI. I haven't tested with BIOS but checked with another SD Card that is slower and has raspbian on it and it's really fast to boot.

kacf, 2024-11-11:
I can confirm as well. Pretty much identical symptoms to Gadget64.

joelpurra, 2025-02-24:
Booting the *same* sd-card, without modification, reliably yields very different loader timings:

- sd-slot: **3+ minutes!**
- usb-reader: 3 seconds

Might be a timeout? Where?

<details>
<summary>systemd-analyze time</summary>

Some [`systemd-analyze time`](https://www.freedesktop.org/software/systemd/man/latest/systemd-analyze.html#systemd-analyze%20time) data points from an rpi4b, [firmware image v1.4.1](https://github.com/pftf/RPi4/releases/tag/v1.41). The below numbers are from cold boots; warm (re)boots yield very similar time spent in the loader (within milliseconds).

Cold boot from the sd-slot:

> Startup finished in 7.166s (firmware) + **3min 6.839s (loader)** + 2.803s (kernel) + 7.456s (initrd) + 6.410s (userspace) = 3min 30.676s

Cold boot from the same sd-card in a USB reader:

> Startup finished in 6.871s (firmware) + **3.731s (loader)** + 2.424s (kernel) + 8.733s (initrd) + 5.438s (userspace) = 27.198s

</details>

---

@Gadget64 wrote:

> Is the card maybe simply running at an extremely low speed at that stage before the kernel has properly configured it? Linux does indeed only report it as `sd high-speed` with `cat /sys/kernel/debug/mmc*/ios` (which should be 25 MB/s) before I've properly applied the device-trees, and only then is it shown as `sd uhs DDR50` (the expected UHS-I speed of 50 MB/s, which is what vanilla RPi OS reports too), although both should be fast enough. Let me know if I can provide any more information!

Do you apply any [`dtoverlay=`/`dtparam=` overlays/tweaks](https://github.com/raspberrypi/firmware/tree/master/boot/overlays) related to the SD-slot?

ea4nz, 2025-03-14:
My NetBSD-10.1 system also takes looong time loading the kernel from the NetBSD bootloader screen.
Regards.

Rua, 2026-01-20:
It looks like the SD card is also reported as `sd high-speed` for me, even though it's a U3 card that should support higher speeds. I have not applied any device trees, just using the mainline kernel as-is.