pftf/RPi4 #264: Synchronous Exception at 0x0000000037101434

einsteinagogo, 2024-11-21:
Synchronous Exception at 0x0000000037101434 

Still getting this issue on microSD cards Raspberry Pi 4 and eMMC CM4, all is working well, and then gets corrupted.

Tried, v1.35, v1.36, v1.37, v1.38

pm4rcin, 2024-11-30:
> Synchronous Exception at 0x0000000037101434
> 
> Still getting this issue on microSD cards Raspberry Pi 4 and eMMC CM4, all is working well, and then gets corrupted.
> 
> Tried, v1.35, v1.36, v1.37, v1.38
> 
> 18m 04s - https://www.youtube.com/watch?v=oezKPXVJ8WM

Do you have fresh GRUB version? Here's a whole thread on that. Basically turns out that there were some problems with shim and GRUB. https://github.com/pftf/RPi4/issues/235

einsteinagogo, 2024-11-30:
using ESXi ARM fling, the fault occurs because of corruption in RPI_EFI.fd, replace the file fixed.

Not sure that is grub related. as there is now grub, as far as I can see.

EmmanuelKasper, 2025-09-10:
Using a Raspberry PI4 with the same error and exactly the same memory addresse  0x0000000037101434 error,
I confirmed reinstalling the pftf UEFI firmware on the boot media solved the boot issue.
Maintainers, should we close the issue ?