pftf/RPi4 #267: NetBSD 10-Stable RaspberryPi 4 and UEFI v.1.39 not working ok

ea4nz, 2024-12-05:
Hello

I have been using successfuly my RPi4 for months with UEFI firmware
v1.37 from https://github.com/pftf/RPi4. It is solid rock. I have
updated to NetBSD-10 Stable from source twice, one around october and
one time more at the end of november by using binary sysupgrade path
and appeared to be working fine.

netbsd-raspa4$ uname -a
NetBSD netbsd-raspa4 10.0_STABLE NetBSD 10.0_STABLE (GENERIC64) #0:
Thu Nov 28 16:37:05 UTC 2024
mkrepro%mkrepro.NetBSD.org@localhost:/usr/src/sys/arch/evbarm/compile/GENERIC64
evbarm

I have just upgraded UEFI firmware to v1.39 by uncompressing it over
/boot and NetBSD does not work well with it. First time I booted into
the system it appeared to work but when I issued "reboot" kernel
paniced. I rebooted again and wanted to do a pkgin update and pkgin
upgrade and in the middle of firefox upgrading it paniced again. It
definitely does not work ok.

So I came back to UEFI  v1.37 and it appears to be working fine again.

Also have noted some strange things that occured not frequently before
I did any firmware upgrade, also when using NetBSD-10.0 RELEASE.
Sometimes after the reboot command the RPI4 did not come to life. I
had to manually remove the power supply twice because it continued to
be stuck and never saw the raspberry image boot screen that usually
comes before the NetBSD loader comes.

Regards.
Ramiro.

ea4nz, 2024-12-18:
Hello, I have just unpacked v1.38 , made a reboot and it appears to work fine. Anyway, I am going to let de RPi4 doing some hard job for severa hours to be sure it works succesfuly.
I will tell the result.

ea4nz, 2024-12-19:
I confirm that v1.38 works just fine.
Regards
Ramiro

ardbiesheuvel, 2025-01-18:
I've fixed this and created a new release v1.40

ea4nz, 2025-01-18:
Thanks so much,

I have upgraded to 1.40, it seems to work fine after one hour. I am going to stress the system doing intense work for several hours to be sure. 

Thanks

ea4nz, 2025-01-18:
Several hours latter it seems to work fine with v1.40 !
Many thanks.
Ramiro

ardbiesheuvel, 2025-01-18:
Thanks for testing!

closed 2025-01-21 by ardbiesheuvel