pftf/RPi4 #268: Windows 10/11 ARM64 KMODE_EXCEPTION_NOT_HANDLED BSOD screen issue

ImNotDevPekka, 2024-12-11:
Hi, yes I know Windows "isn't supported" BUT it could work before, even the GPIO and other stuff besides wifi. So please, get the UEFI work with windows arm versions, alot of people would appreciate it. (using  https://worproject.com/)

pbatard, 2024-12-11:
Windows 11 build 25168 and later require ARMv8.1. The Pi4 CPU is ARMv8.0. UEFI cannot magically fix that.

If you want to run Windows on ARM64, either use a platform with a CPU that uses ARMv8.1 or later, or, if you only have an ARMv8.0 platform, use one of the older version of Windows that support ARMv8.0.

closed 2024-12-11 by pbatard

Marcinoo97, 2024-12-11:
Pete if the person were to use build above 25168 no error would have being shown. It would just froze forever.  Issue here was caused by lates UEFI package and downgrading it helped this person.

ImNotDevPekka, 2024-12-11:
@pbatard No, the issue wasnt with the ARM version at all - I have tried windows 10 20h2 (19041) and windows 11 22h2 (22621) and none of them worked. So yes, I did what @Marcinoo97 said, I downgraded to v1.35 and guess what? I have a fully functioning windows 11 (22621) without any issues.

ImNotDevPekka, 2024-12-11:
https://github.com/user-attachments/assets/709f7306-bd54-4908-9a7a-fde84ce4d94d

pbatard, 2024-12-11:
Well, if you barely provide any details when opening an issue, and don't even indicate the basic of "version x.y works but version x.z" doesn't, please don't complain about the issue being summarily dismissed on account of people having to __guess__ what  you are reporting about.

At any rate, if you want to help with this, then please use git bisect with the edk2/edk2-platforms, to build a bunch of firmwares, test them and then pinpoint the commit that might have introduced a regression. Do not expect somebody else to do that for you.

ImNotDevPekka, 2024-12-11:
As I said, I downgraded to v1.35, and it works now, but the latest doesn't. I dont know about the other versions. Maybe if I have time I'll check. But don't blame me, because I gave all details I expected, I don't have further info.