pftf/RPi4 #269: The Device Tree isnt passed from start4.elf to the UEFI firmware RPI_EFI.fd.

Schiddik, 2024-12-22:
Hello everyone,

currently Iam doing my Masters in Darmstadt, Germany.

Please believe me when I say that i DEEPLY appreciate any answer to my questions!!! Thanks to You in advance!
I will enumerate my questions so that they can be referenced easily.

My task is to deploy Kubernetes and Ceph onto a cluster of 60 Raspberry Pi Compute Blades.
The Compute Blades use the Compute Module 4 (CM4). Most of the Compute Blades dont have persistent storage.

Thus, my task is to boot the CM4s via PXE.
I have already booted the CM4 via the "normal" PXE bootflow.
I will now share my understanding of the "normal" PXE bootflow.
I still have a lot of gaps in my understanding. I would highly appreciate if You could fill these gaps with Your knowledge.

[image]

The initial rootfilesystem was built by Buildroot. It has a custom /init script.
The kernel (vmlinuz.gz) doesnt have an UEFI stub and is generated by the Raspberry PI Imager (Ubuntu 24.04). 
Eventhough somewhere on the internet I read that every Ubuntu kernel has a UEFI stub by default. But more on that later. 

1. My first question is if the L2 cache is located in the GPU or in the ARM chip.
2. How does the second stage bootloader (SSB) (formerly bootcode.bin) know about the onboard NIC? Are the MMIO regions and the interrupt lines and the capabilities compiled into the SSB?
3. How are the files like meta-data and network-config passed into the kernel? 
4. How does the kernel accept this information or these files? Doesnt the kernel need an argument in its entry point for this kind of information? Or are these files copied into the initial rootfilesystem by chance?
5. We all know the standard signatures of programs: int main(argc, argv). Which parameters does the entry point of a NON-UEFI kernel take as arguments? As far as i know one of the needs to be the address of the device tree. Is that correct?
6. What are the program signatures of start4.elf, RPI_EFI.fd, GRUB, a NON-EFI kernel, and a kernel with an EFI stub?
7. What is the fixup4.dat file? What does it contain? When booting via UEFI, the start4.elf "fails to apply fixups". What could be the cause?
8. What is the memory layout of the CM4? Does the start4.elf file need to be loaded to a specific address? Does the kernel need to be loaded to a specific address?
9. How does the kernel know which memory areas it is allowed to use?
10. Is start4.elf unloaded from memory when it passes control to the kernel? How is the stack of the start4.elf program removed?

[image]

In this "normal" PXE boot flow the kernel recognizes the NIC and creates an eth0 interface, which I then use to connect via SSH.

10.1) This question is not related to question 10.  
My current root file system in the "normal" PXE bootflow doesnt have the apt package manager. Thus, iam unable to install new software (eg. the dependencies for kubernetes and ceph). 
Would You have an idea on how to add apt to the root file system? The root filesystem was built with buildroot and they only provice the opkg package manager.
My final goal is to install kubernetes and ceph via ansible. 

**Now coming to the UEFI bootflow:** 

Additionally I was able to boot the Raspberry Pi CM4 (on the Compute Blade) via an UEFI compliant firmware.
To get a hold of the UEFI compliant firmware that would run on the Compute Blade I used this repository: compute-blade-cm4-uefi (https://github.com/uptime-industries/compute-blade-cm4-uefi)
That repository is a fork of the pftf/RPi4 (https://github.com/pftf/RPi4) repository.
Apparently the pftf/RPI4 repository "contains installable builds of the "EDK2 Raspberry Pi 4 UEFI firmware" (https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi4)

[image]

This "EDK2 Raspberry Pi 4 UEFI firmware" repo contains the RPI4.dsc and RPI4.fdf files.
11. What do these files mean and are they standardized in the UEFI specification?

[image]

The generated folder has the /firmware/brcm folder which contains files like brcmfmac43455-sdio.Raspberry. 
12. What are these files used for? When are they used? By whom?

**This is how the directory structure of the UEFI bootflow looks like:**
[image]

The generated config.txt files contains the armstub=RPI_EFI.fd parameter.

13.1 According to my understanding, an armstub is the software that the start4.elf program executes after it is done with all its setup work. 
start4.elf then passes control to the armstub. 
I have read on the internet that an armstub is a small executable that initializes an ARM CPU. 
Some of its tasks are writing specific values to predefined (by the ARM spec) control registers. 
This could result in enabling the branch predictor or the MMU. 
13.2 Is this understanding correct? 
13.3 The input parameters of the armstub are three registers r0,r1,r2. One of these should get the address of the device tree. Is this correct?
[... 65 lines cut ...]
[image]

As You can see in the picture above, the kernel parameters from the cmdline.txt are not passed to the kernel. In my setup the cmdline.txt file contains a kernel parameter.
This can mean many things. start4.elf didnt pass these kernel parameters to RPI_EFI.fd    OR    RPI_EFI.fd didnt pass them to the grub2 bootloader  OR  the grub2 bootloader didnt pass them to the kernel.

27. Does start4.elf pass the kernel parameters to RPI_EFI.fd or does RPI_EFI.fd read those values by itself? And if it reads them by itself, how does it know the location of the cmdline.txt file?
28. What could be the possible cause for the fact, that the kernel parameters are not passed properly?

29. The picture above also shows that there is only the loopback (lo) interface created by the kernel. Does this mean that the kernel didnt find any ethernet devices and thus didnt bother creating an eth0/enp0s3 interface (because he couldnt bind these interfaces to any physical device) ?
30. Does this mean for certain, that the kernel didnt receive a device tree?

I have tried adding this line to the grub.cfg file:   linux (tftp)/vmlinuz   **dtb=/bcm2711-rpi-cm4.dtb** 
31. Would this allow me to put the device tree into the initial root filesystem, which could then be used by the kernel?

**Optional questions:**
32. How does star4.elf pass the address of the device tree to an armstub?
33. Does start4.elf always call an armstub even if the armstub parameter in config.txt isnt specified? If not specified, does start4.elf call a default stub?
34. In the UEFI boot flow, the armstub RPI_EFI.fd downloads the grub2 bootloader, which then downloads the kernel and intird. How would a different kind of armstub know the address of kernel that it needs to execute? Is the address of the kernel standardized and thus doesnt need to be passed as a parameter into the armstub?
35.  How does the second stage bootloader know the MAC of the internal NIC? Does the NIC have persistent storage that stores the NIC? I heard somewhere that the MAC is computed based on the serial number. If thats so, the where is the serial number of the CM4 board stored? Possibly in the EEPROM or the internal ROM in the GPU?
36. How does RPI_EFI.fd know about the NIC? Are the MMIO regions and interrupt numbers compiled into it? 
37. This is a very vague question whose scope is propably way too big: Which memory regions are used by the 1st, 2nd stage bootloader, start4.elf, RPI_EFI.fd, grub, the kernel?
38. How would start4.elf pass control to a “normal” armstub? Is it the responsibility of the armstub to clean the leftover memory that the start4.elf used? How does it know where the start4.elf was loaded to and how much stack space it required?
39. Does the RPI_EFI.fd firmware “claim” all memory? Meaning, that it acts as if all of the SDRAM belongs to it. And does RPI_EFI.fd only keep the region where itself is located (the location where the RPI_EFI.fd firmware was loaded needs to be known beforehand in this case).

Iam DEEPLY grateful for every information I can get! Thank You in advance:)

Xeroxxx, 2025-01-05:
> HUGE problem: The RPI_EFI.fd firmware doesnt recognize the MAC address of the internal NIC. However, it is able to use the NIC. The source address of the outgoing ethernet frames is 00:00:00:00:00:00. 

This problem occurs probably your CM4 EEPROM is to old. Seeing on screenshots its dated 26.04.2022.

This issue has been fixed:
2022-09-02 - Add pieeprom-2022-09-02 - BETA + STABLE
Parse target MAC address in NETCONSOLE property 

https://github.com/raspberrypi/rpi-eeprom/issues/440

Make sure you install the latest EEPROM on all devices.

Beside some other questions I feel there is some basic understanding of DeviceTrees and UEFI missing. As well as parsing the code of grub2 and understand how it works.

Greetings from Frankfurt

Schiddik, 2025-01-06:
@Xeroxxx Thank You for Your answer !!!
I will try to update the EEPROM.

valtzu, 2025-02-26:
The device tree issue could be related to https://github.com/pftf/RPi4/pull/274 ?

ardbiesheuvel, 2025-05-25:
Presumably fixed by #274

closed 2025-05-25 by ardbiesheuvel