pftf/RPi4 #270: Support ubuntu-22.04.5-live-server-arm64

Martin11180, 2024-12-30:
1
Setup for the SD card
```
sudo parted --script /dev/sda mklabel msdos
sudo parted --script /dev/sdd mkpart primary fat32 0% 1GB
sudo mkfs.vfat /dev/sda
sudo fatlabel /dev/sdf1 RPI-UEFI 
sudo mount /dev/sda1 /mnt/data/
sudo unzip RPi4_UEFI_Firmware_v1.39.zip -d /mnt/data/
sudo umount /mnt/data
```

2
and Rufus 
ubuntu-22.04.5-live-server-arm64.iso 
flashen

3
the SD card in Raspbbers
The setup usb stick with Ubuntu on usb port
Connect an empty SSD to USB
Then boot and print esc 
Select USB stick with Ubuntu Boot Manager
The setup starts 
I then do everything and leave everything the same, including the partition
and select the empty SSD disk
The setup starts but stop during Grub install, see picture

[image]

What am I doing wrong?
Unfortunately there are no instructions on how to proceed after the SD card with RPi4_UEFI_Firmware_v1.39 is installed
Whether certain settings need to be made in Ubuntu and with which version

Martin11180, 2024-12-31:
I don't know if the path is right 
I have the iso with Rufus
ubuntu-22.04.5-live-server-arm64.iso
Created with GPT on an SD
Then I copied the RPi4_UEFI_Firmware_v1.38 to the SD card
Then I created an empty SDD disk with GPT partition

I then connected the SD card to the blue USB port using a USB Card Reader Micro SD adapter
And the hard drive also connects to the blue USB port
The Ubuntu setup is running
After the reboot I deleted the SD card
and created an empty partition with GBT 
and then only copied the files from RPi4_UEFI_Firmware_v1.38
and packed into Rasberry cd card slot
And the hard drive and connected to USB2 port
Ubuntu boots

If you do this correctly, you can get it to work without a CD card 
only from SSD?

Martin11180, 2024-12-31:
I connected the new SSD to another Raspberry
and executed the following commands

sudo wget https://github.com/pftf/RPi4/releases/download/v1.38/RPi4_UEFI_Firmware_v1.38.zip
sudo mkdir /mnt/ubununew
sudo mount /dev/sda1 /mnt/ubununew
cd /mnt/ubununew/
sudo unzip RPi4_UEFI_Firmware_v1.38.zip -d /mnt/ubununew
sudo umount /mnt/ubununew

Remove the SD card and boot directly from the SSD

Now I have to see how it works with cloud-init autointsall and whether I have to do it via the ISO or another way

ardbiesheuvel, 2025-02-08:
v1.39 was broken, please try v1.40 instead

closed 2025-02-08 by ardbiesheuvel