pftf/RPi4 #272: Change boot order with efibootmgr  resets after reboot

MichaelKoch11, 2025-02-13:
I tried to adjust the boot order with the program efibootmgr, but it will overwrite the last selected option as the first one.

Why should this work like this? I want to install my operating system on an external HDD via IPXE, which works so far, and then adjust the boot order automatically. (So that my script does not cause any errors, I have added a hold command to test this operation manually first. After that the order is not changed).

Next I manually booted into the correct boot option and after reboot this is now the first operation and it boots into the installed OS again.
Here I tried again to change the boot order so that first network boot and then hard disk is there. After a restart, however, ubuntu is available first and then my ipxe boot option and not as previously specified.

Is there a possibility to set the boot order the way I want it or do I have to think of another solution.

ardbiesheuvel, 2025-02-16:
Could you please explain more clearly
- what you are trying to achieve
- what you are doing
- what the expected result is
- what the actual result is

If `efibootmgr` updates are simply being ignored after a reboot, it might be an issue with the varstore writeback. You can use the EFI shell to create boot options to, using `bcfg boot add <index> <filename> "<label>"`

ardbiesheuvel, 2025-02-17:
> I will use PXE/IPXE to boot and autoinstall to install an OS on a connected HDD. This method works. After the reboot, I want to directly boot into the operating system. To do this, I must change the bootorder in the late-commands of the autoinstall script (changing the bootorder is the first task).

I don't know what this means. Which 'late-commands'? Which autoinstall script?
Are you using `efibootmgr` to set the boot order?

> Later, when the OS is fully booted, a service should automatically check if there are changes to the installed OS. Firstly, it should check for a new version, and if so, change the boot order to IPXE/PXE first and the old OS as a backup. This will install the new OS after the reboot. The UEFI Raspi Bootloader is installed on a Micro SD Card and the HDD is connected via USB.
> 

What is 'the UEFI Raspi Bootloader'?

> When I try to change the boot order automatically, it's the same as before: first PXE/IPXE, then the automatically added UEFI Western Digital, and then UBUNTU. After I select UBUNTU as the boot option manually, it's the first boot option after the reboot, which is fine. However, when I want to reinstall the OS, I cannot change it back to first IPXE/PXE out of the OS, only when I manually select or reorder it directly in the UEFI Menu.

If changes made using `efibootmgr` are not working for you, you should be able to set the boot order from the UEFI Shell. Beyond that, I am not sure what you are asking for, and whether it is related to UEFO

MichaelKoch11, 2025-02-17:
You asked what I was planning to do, so I have outlined it completely.

Question 1:
Yes, the late commands in the autoinstall script.
There I try to change the boot order from PXE/HDD/ etc. to HDD/PXE with the efibootmgr.

Question two:
Perhaps a little misnamed. I meant the Raspberry Pi 4 UEFI firmware images provided in the repository.

Specifically, it's about changing the boot order in the UEFI firmware and making it as automated as possible. Which is why I am looking for a way to automatically change the boot order from the running operating system or after the operating system installation. Without having to make manual changes.

Translated with DeepL.com (free version)

MichaelKoch11, 2025-02-17:
You asked what I was planning to do, so I have outlined it completely.

Question 1:
Yes, the late commands in the autoinstall script.
There I try to change the boot order from PXE/HDD/ etc. to HDD/PXE with the efibootmgr.

Question two:
Perhaps a little misnamed. I meant the Raspberry Pi 4 UEFI firmware images provided in the repository.

Specifically, it's about changing the boot order in the UEFI firmware and making it as automated as possible. Which is why I am looking for a way to automatically change the boot order from the running operating system or after the operating system installation. Without having to make manual changes.

ardbiesheuvel, 2025-02-17:
> Specifically, it's about changing the boot order in the UEFI firmware and making it as automated as possible. Which is why I am looking for a way to automatically change the boot order from the [...]

OK, so this is a generic Linux+UEFI related question then? And you are not reporting a defect in the RPI4 images?

MichaelKoch11, 2025-02-17:
I use a Raspberry Pi 4B. And I wanted to change the boot sequence from Linux. In generic Linux+UEFI this works with efibootmgr for automatic boot order adjustment.
Now the question was whether this should and can also work here or whether this is a bug in the firmware?
Or whether a different approach must be used for this firmware.

ardbiesheuvel, 2025-02-17:
No, sadly, `efibootmgr` does not work quite in the same way on RPi4B. But this is a hardware limitation, not a bug in the firmware.

closed 2025-02-17 by ardbiesheuvel