pftf/RPi4 #275: 1.41 does not boot

ea4nz, 2025-03-12:
Hello,

I have unpacked v1.41 in my RPi4 and it did not boot. I made it twice with no luck. I see the coloured screen but no the raspberry. Green light flashed 7 times repeatedly. I went back to 1.40 and worked. I do not know if I made something wrong,  it seems easy...

ea4nz, 2025-03-12:
I have started this time  with a clean SD and created and formated a unique new msdos partition using fdisk under a NetBSD amd64 system. I extracted v1.41 firmware and seems to boot fine in the RPI4 and reach  the UEFI screen. Sorry for the noise.

I still do not understand why it did not boot when extracted over the NetBSD-current  aarch64 image file on the SD. I will repeat again the experiment to check that.

Regards.

ea4nz, 2025-03-12:
I have  just downloaded

https://nycdn.netbsd.org/pub/NetBSD-daily/HEAD/202503110400Z/evbarm-aarch64/binary/gzimg/arm64.img.gz

unzipped it and flashed in the SD with dd command, then mounted the msdos partition and copied RPi4_UEFI_Firmware_v1.41.zip. Extracted over the NetBSD aarch64 files and fired up.

It got stuck at the multi-color screen, not passing beyond it. I do not understand anything.

Regards.

ea4nz, 2025-03-12:
I confirm that unpacking old v1.40 over the same NetBSD-current SD image works. I do not understand what happens.
Regards.

ea4nz, 2025-03-14:
Hello, I have discovered that using NetBSD-current arm64mbr.img (MBR partition scheme) instead arm64.img  (GPT partition scheme) works perfectly.

My EEPROM versión: 

root@raspa4:~# rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Tue 11 Feb 17:00:13 UTC 2025 (1739293213)
    LATEST: Tue 11 Feb 17:00:13 UTC 2025 (1739293213)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2711/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0

Regards

pm4rcin, 2025-03-16:
I have the exact same EEPROM firmware version. I've tried changing values to the ones from https://github.com/pftf/RPi4/pull/274 but no luck.