pftf/RPi4 #278: How to change 3G limit?

Reyhn3, 2025-05-14:
The documentation mentions you can use the Device Manager to change the 3G limit. I'm running my RPi headless and I don't even have a display cable for the RPi 4, making it impossible for me to change it in a boot UI.

Is there any way I can disable the 3G limit in the firmware before I flash it to a SD-card?

For instance, a parameter in the config.txt or somewhere?
As you can see, I have no idea of what I'm talking about - this is way beyond me. All I want to do is run Fedora CoreOS on my RPi4 with 8 GB RAM.

markafarrell, 2025-05-19:
@Reyhn3  you can use `virt-fw-vars` from https://pypi.org/project/virt-firmware/

```bash
cat <<EOF > fw-vars.json
{
    "version": 2,
    "variables": [
        {
            "name": "RamLimitTo3GB",
            "guid": "cd7cc258-31db-22e6-9f22-63b0b8eed6b5",
            "data": "00000000"
        }
    ]
}
EOF

sudo virt-fw-vars \
    --input RPI_EFI.fd \
    --output RPI_EFI.fd \
    --set-json fw-vars.json
```

Reyhn3, 2025-06-13:
This is the output from the **original** file:
```bash
❯ virt-fw-vars --input RPI_EFI.fd.bak --print
INFO: reading raw edk2 varstore from RPI_EFI.fd.bak
INFO: var store range: 0x3b0064 -> 0x3be000
```

This is the output from the **modified** file:
```bash
❯ virt-fw-vars --input RPI_EFI.fd --print
INFO: reading raw edk2 varstore from RPI_EFI.fd
INFO: var store range: 0x3b0064 -> 0x3be000
RamLimitTo3GB       : dword: 0x00000000
```

This is the output from the file on the **SD-card**:
```bash
❯ virt-fw-vars --input /mnt/rpi-sd/RPI_EFI.fd --print
INFO: reading raw edk2 varstore from /mnt/rpi-sd/RPI_EFI.fd
INFO: var store range: 0x3b0064 -> 0x3be000
AssetTag            : blob: 66 bytes
Attempt 1           : blob: 1049 bytes
Attempt 2           : blob: 1049 bytes
Attempt 3           : blob: 1049 bytes
Attempt 4           : blob: 1049 bytes
Attempt 5           : blob: 1049 bytes
Attempt 6           : blob: 1049 bytes
Attempt 7           : blob: 1049 bytes
Attempt 8           : blob: 1049 bytes
Boot0000            : boot entry: title="UiApp" devpath=FvName(9a15aa37-d555-4a4e-b541-86391ff68164)/FvFileName(462caa21-7614-4503-836e-8ab6f4662331)
Boot0001            : boot entry: title="UEFI Misc Device" devpath=VendorHW(100c2cfa-b586-4198-9b4c-1683d195b1da) optdata=4eac0881119f594d850ee21a522c59b2
Boot0002            : boot entry: title="UEFI PXEv4 (MAC:DCA632ED4555)" devpath=MAC()/IPv4() optdata=4eac0881119f594d850ee21a522c59b2
Boot0003            : boot entry: title="UEFI PXEv6 (MAC:DCA632ED4555)" devpath=MAC()/IPv6() optdata=4eac0881119f594d850ee21a522c59b2
Boot0004            : boot entry: title="UEFI HTTPv4 (MAC:DCA632ED4555)" devpath=MAC()/IPv4()/URI() optdata=4eac0881119f594d850ee21a522c59b2
Boot0005            : boot entry: title="UEFI HTTPv6 (MAC:DCA632ED4555)" devpath=MAC()/IPv6()/URI() optdata=4eac0881119f594d850ee21a522c59b2
Boot0006            : boot entry: title="UEFI Shell" devpath=FvName(9a15aa37-d555-4a4e-b541-86391ff68164)/FvFileName(7c04a583-9e3e-4f1c-ad65-e05268d0b4d1)
BootDiscoveryPolicy : dword: 0x00000002
BootDiscoveryPolicyOld: dword: 0x00000002
BootOrder           : boot order: 0000, 0001, 0002, 0003, 0004, 0005, 0006
ClientId            : blob: 20 bytes
ConIn               : devpath: Msg(subtype=0xf)
ConOut              : devpath: VendorHW(d3987d4b-971a-435f-8caf-4967eb627241)/Msg(subtype=0xe)/Msg(subtype=0xa)
ConsolePref         : dword: 0x00000000
CpuClock            : dword: 0x00000001
CustomCpuClock      : dword: 0x00000708
CustomMode          : bool: off
DCA632ED4555        : blob: 60 bytes
DebugEnableJTAG     : dword: 0x00000000
DisplayEnableSShot  : dword: 0x00000001
DisplayEnableScaledVModes: byte: 0x20
ErrOut              : devpath: VendorHW(d3987d4b-971a-435f-8caf-4967eb627241)/Msg(subtype=0xe)/Msg(subtype=0xa)
FanOnGpio           : dword: 0x00000000
FanTemp             : dword: 0x0000003c
InitialAttemptOrder : qword: 0x0807060504030201
Key0000             : blob: 14 bytes
Key0001             : blob: 14 bytes
Lang                : ascii: "eng"
MTC                 : dword: 0x00000001
MmcDisableMulti     : dword: 0x00000000
MmcEnableDma        : dword: 0x00000001
MmcForce1Bit        : dword: 0x00000000
MmcForceDefaultSpeed: dword: 0x00000000
MmcSdDefaultSpeedMHz: dword: 0x00000019
MmcSdHighSpeedMHz   : dword: 0x00000032
PlatformLang        : ascii: "en-US"
RamLimitTo3GB       : dword: 0x00000000
RamMoreThan3GB      : dword: 0x00000001
RtcDaylight         : byte: 0x00
RtcEpochSeconds     : qword: 0x000000006832e08e
RtcTimeZone         : word: 0x07ff
SdIsArasan          : dword: 0x00000000
SystemTableMode     : dword: 0x00000000
Timeout             : word: 0x0005
VarErrorFlag        : byte: 0xff
VendorKeysNv        : byte: 0x01
XhciPci             : dword: 0x00000000
XhciReload          : dword: 0x00000000
_NDL                : blob: 41 bytes
certdb              : dword: 0x00000004
```

I'm just guessing here, but it looks like the modified and the SD-card files are both correct.
But still, the memory is limited on the RPi:

```bash
$ free -h
               total        used        free      shared  buff/cache   available
Mem:           2.8Gi       263Mi       2.4Gi       720Ki       251Mi       2.5Gi
Swap:             0B          0B          0B
```

Am I doing something wrong?

markafarrell, 2025-06-13:
Not sure if it helps but this is what mine looks like

```bash
$virt-fw-vars --input RPI_EFI.fd --print
INFO: reading raw edk2 varstore from RPI_EFI.fd
INFO: var store range: 0x3b0064 -> 0x3be000
000000000000        : blob: 60 bytes
AssetTag            : blob: 66 bytes
Attempt 1           : blob: 1049 bytes
Attempt 2           : blob: 1049 bytes
Attempt 3           : blob: 1049 bytes
Attempt 4           : blob: 1049 bytes
Attempt 5           : blob: 1049 bytes
Attempt 6           : blob: 1049 bytes
Attempt 7           : blob: 1049 bytes
Attempt 8           : blob: 1049 bytes
Boot0000            : boot entry: title="UiApp" devpath=FvName(9a15aa37-d555-4a4e-b541-86391ff68164)/FvFileName(462caa21-7614-4503-836e-8ab6f4662331)
Boot0001            : boot entry: title="UEFI ASMT USB 3.0 Destop HD EP0 Product string 0000000000A1" devpath=ACPI(hid=0xa0841d0,uid=0x0)/PCI(dev=00:0)/PCI(dev=00:0)/USB(port=1) optdata=4eac0881119f594d850ee21a522c59b2
Boot0002            : boot entry: title="UEFI PXEv4 (MAC:000000000000)" devpath=MAC()/IPv4() optdata=4eac0881119f594d850ee21a522c59b2
Boot0003            : boot entry: title="UEFI PXEv6 (MAC:000000000000)" devpath=MAC()/IPv6() optdata=4eac0881119f594d850ee21a522c59b2
Boot0004            : boot entry: title="UEFI Shell" devpath=FvName(9a15aa37-d555-4a4e-b541-86391ff68164)/FvFileName(7c04a583-9e3e-4f1c-ad65-e05268d0b4d1)
BootDiscoveryPolicy : dword: 0x00000002
BootDiscoveryPolicyOld: dword: 0x00000002
BootOrder           : boot order: 0000, 0001, 0002, 0003, 0004
ClientId            : blob: 20 bytes
ConIn               : devpath: Msg(subtype=0xf)
ConOut              : devpath: VendorHW(d3987d4b-971a-435f-8caf-4967eb627241)/Msg(subtype=0xe)/Msg(subtype=0xa)
ConsolePref         : dword: 0x00000000
CpuClock            : dword: 0x00000001
CustomCpuClock      : dword: 0x00000708
CustomMode          : bool: off
DebugEnableJTAG     : dword: 0x00000000
DisplayEnableSShot  : dword: 0x00000001
DisplayEnableScaledVModes: byte: 0x20
ErrOut              : devpath: VendorHW(d3987d4b-971a-435f-8caf-4967eb627241)/Msg(subtype=0xe)/Msg(subtype=0xa)
FanOnGpio           : dword: 0x00000000
FanTemp             : dword: 0x0000003c
InitialAttemptOrder : qword: 0x0807060504030201
Key0000             : blob: 14 bytes
Key0001             : blob: 14 bytes
Lang                : ascii: "eng"
MTC                 : dword: 0x00000005
MemoryTypeInformation: blob: 48 bytes
MmcDisableMulti     : dword: 0x00000000
MmcEnableDma        : dword: 0x00000001
MmcForce1Bit        : dword: 0x00000000
MmcForceDefaultSpeed: dword: 0x00000000
MmcSdDefaultSpeedMHz: dword: 0x00000019
MmcSdHighSpeedMHz   : dword: 0x00000032
PlatformLang        : ascii: "en-US"
RamLimitTo3GB       : dword: 0x00000000
RamMoreThan3GB      : dword: 0x00000001
ResetDelay          : dword: 0x003567e0
RtcDaylight         : byte: 0x00
RtcEpochSeconds     : qword: 0x0000000067b8a2a0
RtcTimeZone         : word: 0x07ff
SdIsArasan          : dword: 0x00000000
SystemTableMode     : dword: 0x00000001
Timeout             : word: 0x0005
VarErrorFlag        : byte: 0xff
VendorKeysNv        : byte: 0x01
XhciPci             : dword: 0x00000000
XhciReload          : dword: 0x00000000
_NDL                : blob: 41 bytes
certdb              : dword: 0x00000004
```

```bash
$ free -h
               total        used        free      shared  buff/cache   available
Mem:           7.6Gi       5.8Gi       141Mi        21Mi       2.0Gi       1.8Gi
```

Reyhn3, 2025-06-26:
I got this to work! I even got a HDMI-cable so I could verify visually with a display.

My goal was to get **Fedora CoreOS** running on my 8 GB **Raspberry Pi 4** Model B with all memory available, and it was a bit easier than what I tried in previous posts. This is what I did:

Based on [Fedora's provisioning](https://docs.fedoraproject.org/en-US/fedora-coreos/provisioning-raspberry-pi4/#_edk2_combined_disk_mode_alternate_machine_disk_preparation), I created a EDK2 combined disk from my laptop.

1. I created an ignition file using Butane [as described](https://docs.fedoraproject.org/en-US/fedora-coreos/producing-ign/) in the CoreOS documentation (the file is named `fcos.ign` below).
2. I used `lsblk` to identify my SD-card as `/dev/mmcblk0`.
3. Download the Fedora image:
```bash
coreos-installer download -s stable -a aarch64
```
4. Write the CoreOS to the SD-card:
```bash
sudo coreos-installer install \
	-a aarch64 \
	-f fedora-coreos-42.20250609.3.0-metal.aarch64.raw.xz \
	-i fcos.ign \
	/dev/mmcblk0
```
5. Instead of writing the EDK2 firmware files [as instructed](https://docs.fedoraproject.org/en-US/fedora-coreos/provisioning-raspberry-pi4/#_edk2_combined_disk_mode_alternate_machine_disk_preparation), I combined it with the tip I got from @markafarrell  like this:
```bash

VERSION=1.42

FCOSEFIPARTITION=$(lsblk $FCOSDISK -J -oLABEL,PATH  | jq -r '.blockdevices[] | select(.label == "EFI-SYSTEM")'.path)
mkdir /tmp/FCOSEFIpart
sudo mount $FCOSEFIPARTITION /tmp/FCOSEFIpart -o dmask=000,fmask=111
pushd /tmp/FCOSEFIpart
sudo curl -LO https://github.com/pftf/RPi4/releases/download/v${VERSION}/RPi4_UEFI_Firmware_v${VERSION}.zip
sudo unzip RPi4_UEFI_Firmware_v${VERSION}.zip

virt-fw-vars --input RPI_EFI.fd --print

echo -e "\e[1;30;33m$(sha256sum RPI_EFI.fd)\e[0m (before)"
virt-fw-vars \
    --input RPI_EFI.fd \
    --output RPI_EFI.fd \
    --set-json <(cat <<EOF
{
    "version": 2,
    "variables": [
        {
            "name": "RamLimitTo3GB",
            "guid": "cd7cc258-31db-22e6-9f22-63b0b8eed6b5",
            "data": "00000000"
        },
        {
            "name": "SystemTableMode",
            "guid": "cd7cc258-31db-22e6-9f22-63b0b8eed6b5",
            "data": "02000000"
        }
    ]
}
EOF
)
echo -e "\e[1;30;33m$(sha256sum RPI_EFI.fd)\e[0m (after)"

virt-fw-vars --input RPI_EFI.fd --print

sudo rm RPi4_UEFI_Firmware_v${VERSION}.zip
popd
sudo umount /tmp/FCOSEFIpart
```
Note that I both [change the 3GB limit](https://docs.fedoraproject.org/en-US/fedora-coreos/provisioning-raspberry-pi4/#_edk2_firmware_changing_the_3g_limit) and set the [GPIO via DeviceTree](https://docs.fedoraproject.org/en-US/fedora-coreos/provisioning-raspberry-pi4/#_edk2_firmware_gpio_via_devicetree) at the same time.

6. Put the SD-card in the RPi and boot. It takes a long time (very long, maybe 20 minutes or so).
7. When I SSH into the RPi, I can run the following commands to see that I indeed have 8 GB RAM and GPIO devices:
```bash
free -h
ls /proc/device-tree/ | wc -l
ls /dev/gpiochip*
```
8. I rebooted the RPi and pressed ESC to enter the BIOS in which I could confirm that the 3G-limit was disabled and the System Table is set to DeviceTree.
9. Done!

**Note:** 
Notice how the `SystemTableMode` data value is `02000000`. I don't know why the digits have to be in reverse order, but this works. According to the documentation on the [EDK2-platform](https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi4#configuration-settings), it should be `0x00000002`. If I tried to use `"data:" "00000002"`, the value printed by `virt-fw-vars --input RPI_EFI.fd --print` became `0x0200000`. 🤷 

Thank you for all your help, and hope this helps someone else!